MrYsLab/pymata-aio #5: Autodetection fails on os x

bitcraft, 2015-07-21:
I suppose this would be a matter of adding 'usbmodemXXX' or 'usbserialXXX' to the of names to check.

https://www.arduino.cc/en/guide/macOSX#toc8

MrYsLab, 2015-07-21:
I will add both usbmodem and usbserial to the list. I don't own os x machine, so I won't be able to test if it is successful or not, but will let you know when it is ready, and if you can verify that it works, that would be great.

bitcraft, 2015-07-21:
No problem, I'd be happy to test it for you.

MrYsLab, 2015-07-24:
Added OS X port strings in release 1.3

bitcraft, 2015-07-27:
It still fails on OS X.  here is a regular expression that works for me:
(cu|tty)\.usb(modem|serial)\d{3}

The trick is that there is a three digit number after 'usbserial' or 'usbmodem'  so far it seems to just be 641, but that may change on systems with many usb-serial devices.

MrYsLab, 2015-07-27:
Thanks, I am going to have think about this one a little, since the code currently does a hard comparison and not regex. I will see if I can get a full list of possibilities. The Leonardo as well as other 32u4 devices has a habit of "port jumping" when pulling out the USB cable and plugging it back in and for windows and linux it seems to work well and I hope I can do the same of OS X.

MrYsLab, 2015-07-27:
@bitcraft From looking at https://github.com/voodootikigod/node-serialport it appears that only tty. ports are being supported, so I am going to do the same here. I will get a list of serial ports, and sort all of the /dev/tty.usb* ports.  I will then take the first port in the sorted list and use that. Does this seem to make sense from an OS X perspective (that is taking the first from a sorted list)?

bitcraft, 2015-07-27:
That should be ok.  One consideration is that the bluetooth modem that is included on MacBooks is listed as /dev/tty.Bluetooth* (there are two), so you will need the 'usb*' when searching....(no shortcuts!)  It would be better to check the name or use a regular expression to pick out the right device, but the wildcard grep/search would work well enough, imo.

MrYsLab, 2015-07-27:
@bitcraft Could you please try running the following program https://gist.github.com/MrYsLab/97b7b0764da4ea33df99 and tell me:
1. If the full and correct port designations come back.
2. If the first port in the list is the "Arduino" port. (Perhaps I don't need to sort).

Thanks

MrYsLab, 2015-08-01:
Closed with v1.4

closed 2015-08-01 by MrYsLab