MrYsLab/pymata-aio #6: Cannot start a board during asyncio loop

bitcraft, 2015-07-21:
Currently a RuntimeError is raised when attempting to start a board when the event loop is already running.  Will it be possible to start a board while when an event loop has already started?

MrYsLab, 2015-07-21:
Could you please post the error here? It will help me understand your problem better.

bitcraft, 2015-07-21:
I have a fix for this.  It will turn Board.start() into a coroutine so it can be yield from'd like anything else.

```
Traceback (most recent call last):
  File "/Users/leif/PycharmProjects/tailor/tailor/apps/service/__main__.py", line 16, in <module>
    app.run()
  File "/Users/leif/PycharmProjects/tailor/tailor/apps/service/service.py", line 46, in run
    loop.run_until_complete(session.start(template_graph_root))
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/base_events.py", line 316, in run_until_complete
    return future.result()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/futures.py", line 275, in result
    raise self._exception
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/tasks.py", line 238, in _step
    result = next(coro)
  File "/Users/leif/PycharmProjects/tailor/tailor/apps/service/service.py", line 79, in start
    yield from wait_for_trigger()
  File "/Users/leif/PycharmProjects/tailor/tailor/hardware/arduino.py", line 9, in wait_for_trigger
    yield from board.start()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/pymata_aio/pymata_core.py", line 161, in start
    report = self.loop.run_until_complete(self.get_analog_map())
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/base_events.py", line 304, in run_until_complete
    self.run_forever()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/base_events.py", line 271, in run_forever
    raise RuntimeError('Event loop is running.')
RuntimeError: Event loop is running.
```

MrYsLab, 2015-07-22:
Could you please provide a short program that demonstrates this problem. Turning start into a coroutine creates some problems on its own and I want to make sure that any fix I come up with solves all issues. Thanks.

bitcraft, 2015-07-22:
```python
from pymata_aio.pymata_core import PymataCore
import asyncio

if __name__ == "__main__":
    board = PymataCore()
    loop = asyncio.get_event_loop()
    loop.call_soon(board.start)
    loop.run_forever()
```

And the traceback:
```
Exception in callback PymataCore.start()
handle: <Handle PymataCore.start()>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/pymata_aio/pymata_core.py", line 161, in start
    report = self.loop.run_until_complete(self.get_analog_map())
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/base_events.py", line 304, in run_until_complete
    self.run_forever()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/asyncio/base_events.py", line 271, in run_forever
    raise RuntimeError('Event loop is running.')
RuntimeError: Event loop is running.
Task was destroyed but it is pending!
task: <Task pending coro=<_command_dispatcher() running at /Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/pymata_aio/pymata_core.py:787> wait_for=<Future pending cb=[Task._wakeup()]>>
```

MrYsLab, 2015-07-23:
Would the following example solve your problem?
```
from pymata_aio.pymata_core import PymataCore
import asyncio

@asyncio.coroutine
def get_report():
    report = yield from board.get_analog_map()
    print (report)

if __name__ == "__main__":
    board = PymataCore()
    board.start()
    loop = asyncio.get_event_loop()
    loop.run_until_complete(get_report())
    loop.run_forever()

```

bitcraft, 2015-07-23:
Yes, this works, but it doesn't really help the issue.  Its pretty unusual for a asyncio-aware class to create its own event loop, as Board.start does.  The code you just replied with still has the issue where you are forced to start the board before the rest of the program.  While I can see it works well in your examples, it creates hurdles for other developers to use.  

In my application, arduino support is optional and may or may not be started while the program is running.  With the way Board.start is programed now, it isn't possible to to load/unload the arduino at runtime, because it interferes with the event loop.

I have a fork that you can take a look at.  The changes I made to Board.start allow the board to be configured at any time.

https://github.com/bitcraft/pymata-aio/commit/b92d3fe9bfd2e5ff5044e21b3af5d03feb263420
https://github.com/bitcraft/pymata-aio/blob/master/pymata_aio/pymata_core.py#L138

bitcraft, 2015-07-23:
Maybe, you could keep the start method, which implements the old behavior of starting an event loop, but move all the board init code (get_analog_map and friends) to __init__?  So async people like me can just skip using start() and all others can use it to get an event loop.

MrYsLab, 2015-07-23:
I will investigate this for the future, but since the code works as designed, I will close this issue. Please feel free to comment or reopen.

MrYsLab, 2015-08-01:
@bitcraft I have revisited this issue, and am thinking of including an asyncio version of start(). Here is  an example of using this. Please let me know if this would meet your needs.

```
import asyncio

from pymata_aio.pymata_core import PymataCore
from pymata_aio.constants import Constants

def data_val(data):
    """
    This is the callback method used to report data from the i2c device
    @param data: Data returned from the device
    @return: No return value
    """
    print(str(data))

@asyncio.coroutine
def i2c_play(board):
    """
    Read the who_am_i register of the device
    """
    device_info = {'DEVICE_ADDRESS': 0x1d, 'WHO_AM_I': 0x0d}

    addr = device_info['DEVICE_ADDRESS']
    register = device_info['WHO_AM_I']
    num_bytes_returned = 1

    # configure Firmata for i2c
    yield from board.i2c_config()

    # read the WHO AM I register of the MMA8452Q Accelerometer, expect 1 byte back
    yield from board.i2c_read_request(addr, register, num_bytes_returned,
                                      Constants.I2C_READ | Constants.I2C_END_TX_MASK,
                                      data_val)

    # give i2c device time to respond
    yield from asyncio.sleep(1)

# instantiate PymataCore
my_board = PymataCore(2)

loop = asyncio.get_event_loop()

# execute the asyncio version of start
loop.run_until_complete(my_board.start_aio())

# get to work
loop.run_until_complete(i2c_play(my_board))
```

bitcraft, 2015-08-10:
Thanks for thinking about me.  I can say right now that your changes don't really help unless the developer can manage multiple event loops, perhaps in threads?.  The core issue is that the several of the pymata_aio methods involved (I've gone through them all!) expect to have total control of the event loop.  In typical asyncio programs (that i've worked with), it is accepted that all participants of the asyncio 'framework' will work with coroutines and not take control of the loop or kill it.  There are several offenders in pymata_aio that will close the event loop, or expect to get total control, and that is basically what keeps pymata_aio from really working well in the asyncio world, in my opinion (i do appreciate the work, and it does work well within your expecatations....don't get me wrong).
TL;DR no those changes don't affect the issue; starting a board is still only possible before the event loop is started.

MrYsLab, 2015-08-10:
Thanks for the input. I will take a look at what it will take to make pymata "loop agnostic" for a future release. Stay tuned :-).

I will open up a new issue to keep it on the burner and when I release 1.5, will close this issue.

MrYsLab, 2015-08-15:
Closing this issue.

closed 2015-08-15 by MrYsLab