MrYsLab/pymata-aio #13: Create a loop agnostic version of pymata

MrYsLab, 2015-08-10:
See https://github.com/MrYsLab/pymata-aio/issues/6 for details.

bitcraft, 2015-09-30:
With version 2.x breaking backwards compatibility with 1.x series, could we address the 'pymata-aio breaking the event loop' issue?  In order to make pymata-aio cooperate with an event loop, many changes will need to be made and it will break old code.  I've been maintaining a pymata-aio fork that cooperates with the asyncio event loop, and I would be happy to help bring better compatibility with python 3.5 (or 3.4!).

MrYsLab, 2015-09-30:
@bitcraft Could you please provide a pointer to your fork? I would like to get a better understanding of what is involved. When you say "it will break old code", does that mean the breaking the existing the library itself or existing applications developed with pymata_aio or both?

FYI - any applications developed with version 1.x of the library should work with the version 2 (Pymata 3.5) without modification. The Python folks have not removed (at least not yet) the Python 3.4.3 syntax for asyncio.

bitcraft, 2015-09-30:
I'll get the code up today. When I say "breaking changes" I mean changes
for folks who are still on 3.4.

MrYsLab, 2015-12-01:
I am closing this issue since the code meets since I am unlikely to implement the changes necessary for this.

closed 2015-12-01 by MrYsLab