MrYsLab/pymata-aio #14: pymata3 losing messages if many are sent at once

fisherds, 2015-08-13:
If many messages are sent at once using pymata3 the later messages are not sent.  For example I'm doing 2 digit writes then an analog write (done twice).  I believe it's the first digital write after the analog write that never works.  I'm getting around this by resending them in the code below, but obviously that isn't good. :)

Video showing the code below running.
http://youtu.be/J5gNZM6pQJA
Note I'm using an XBee in the video, but the issue is not with the XBee.  Exact same behavior with the direct USB cable connection.

    #!/usr/bin/env python3
    """
    Copyright (c) 2015 Alan Yorinks All rights reserved.
    This program is free software; you can redistribute it and/or
    modify it under the terms of the GNU  General Public
    License as published by the Free Software Foundation; either
    version 3 of the License, or (at your option) any later version.
    This library is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
    General Public License for more details.
    You should have received a copy of the GNU General Public
    License along with this library; if not, write to the Free Software
    Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
    """
    
    import time
    import sys
    import signal
    
    from pymata_aio.pymata3 import PyMata3
    from pymata_aio.constants import Constants
    
    BOARD_LED = 13
    
    # RedBot motor pins (from RedBot.h)
    L_CTRL_1 = 2
    L_CTRL_2 = 4
    PWM_L = 5
    
    R_CTRL_1 = 7
    R_CTRL_2 = 8
    PWM_R = 6
    
    board = PyMata3()
    
    def signal_handler(sig, frame):
        print('\nYou pressed Ctrl+C')
        if board is not None:
            board.shutdown()
        sys.exit(0)
    
    def setup():
        """Setup pins"""
        signal.signal(signal.SIGINT, signal_handler) # Handle Ctrl-C to shutdown
        print("Simple drive")
        board.set_pin_mode(BOARD_LED, Constants.OUTPUT)
        board.set_pin_mode(L_CTRL_1, Constants.OUTPUT)
        board.set_pin_mode(L_CTRL_2, Constants.OUTPUT)
        board.set_pin_mode(PWM_L, Constants.PWM)
        board.set_pin_mode(R_CTRL_1, Constants.OUTPUT)
        board.set_pin_mode(R_CTRL_2, Constants.OUTPUT)
        board.set_pin_mode(PWM_R, Constants.PWM)
    
    def loop(use_resend_fallback_mechanism):
        if use_resend_fallback_mechanism:
            print("USING resend fallback mechanism.")
        else:
            print("Only send each command once")
    
        print("Straight")
        board.digital_write(BOARD_LED, 1)
        board.digital_write(R_CTRL_1, 1)
        board.digital_write(R_CTRL_2, 0)
        board.analog_write(PWM_R, 245)
        board.digital_write(L_CTRL_1, 1)
        board.digital_write(L_CTRL_2, 0)
        board.analog_write(PWM_L, 245)
    
        if use_resend_fallback_mechanism:
            time.sleep(0.1)
            # Resend to be sure messages sent second arrived.
            board.digital_write(L_CTRL_1, 1)
            board.digital_write(L_CTRL_2, 0)
            board.analog_write(PWM_L, 245)
    
        time.sleep(2.0)
    
        print("Left!")
        board.digital_write(BOARD_LED, 0)
        board.digital_write(R_CTRL_1, 0)
        board.digital_write(R_CTRL_2, 1)
        board.analog_write(PWM_R, 245)
        board.digital_write(L_CTRL_1, 1)
        board.digital_write(L_CTRL_2, 0)
        board.analog_write(PWM_L, 245)
        time.sleep(2.0)
    
        print("Right")
        board.digital_write(BOARD_LED, 1)
        board.digital_write(R_CTRL_1, 1)
        board.digital_write(R_CTRL_2, 0)
        board.analog_write(PWM_R, 245)
        board.digital_write(L_CTRL_1, 0)
        board.digital_write(L_CTRL_2, 1)
        board.analog_write(PWM_L, 245)
        time.sleep(2.0)
    
    
        print("Stop")
        board.digital_write(BOARD_LED, 0)
        board.digital_write(R_CTRL_1, 1)
        board.digital_write(R_CTRL_2, 0)
        board.analog_write(PWM_R, 0)
        board.digital_write(L_CTRL_1, 1)
        board.digital_write(L_CTRL_2, 0)
        board.analog_write(PWM_L, 0)
    
        if use_resend_fallback_mechanism:
            time.sleep(0.1)
            # Resend to be sure messages sent second arrived.
            board.digital_write(L_CTRL_1, 1)
            board.digital_write(L_CTRL_2, 0)
            board.analog_write(PWM_L, 0)
    
        time.sleep(5.0)
    
    
    if __name__ == "__main__":
      setup()
      use_resend_fallback_mechanism = True
      while True:
        loop(use_resend_fallback_mechanism)
        use_resend_fallback_mechanism = not use_resend_fallback_mechanism

MrYsLab, 2015-08-13:
@fisherds Please try replacing all of the time.sleep calls with board.sleep. For example:
```
time.sleep(5.0)
```
becomes
```
board.sleep(5.0)
```

MrYsLab, 2015-08-13:
Should not have closed it yet.

fisherds, 2015-08-14:
Yup that worked.  The revised code is below (which is successful). There was a side effect though that I'll ask about in a new bug. :)

		#!/usr/bin/env python3
		"""
		Copyright (c) 2015 Alan Yorinks All rights reserved.
		This program is free software; you can redistribute it and/or
		modify it under the terms of the GNU  General Public
		License as published by the Free Software Foundation; either
		version 3 of the License, or (at your option) any later version.
		This library is distributed in the hope that it will be useful,
		but WITHOUT ANY WARRANTY; without even the implied warranty of
		MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
		General Public License for more details.
		You should have received a copy of the GNU General Public
		License along with this library; if not, write to the Free Software
		Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
		"""

		import sys
		import signal
		import time

		from pymata_aio.pymata3 import PyMata3
		from pymata_aio.constants import Constants

		BOARD_LED = 13

		# RedBot motor pins (from RedBot.h)
		L_CTRL_1 = 2
		L_CTRL_2 = 4
		PWM_L = 5

		R_CTRL_1 = 7
		R_CTRL_2 = 8
		PWM_R = 6

		board = PyMata3()

		def signal_handler(sig, frame):
		    print('\nYou pressed Ctrl+C')
		    if board is not None:
		        board.shutdown()
		    sys.exit(0)
	
		signal.signal(signal.SIGINT, signal_handler) # Handle Ctrl-C to shutdown

		def setup():
		    """Setup pins"""
		    print("Simple drive")
		    board.set_pin_mode(BOARD_LED, Constants.OUTPUT)
		    board.set_pin_mode(L_CTRL_1, Constants.OUTPUT)
		    board.set_pin_mode(L_CTRL_2, Constants.OUTPUT)
		    board.set_pin_mode(PWM_L, Constants.PWM)
		    board.set_pin_mode(R_CTRL_1, Constants.OUTPUT)
		    board.set_pin_mode(R_CTRL_2, Constants.OUTPUT)
		    board.set_pin_mode(PWM_R, Constants.PWM)

		def loop():
		    print("Straight")
		    board.digital_write(BOARD_LED, 1)
		    board.digital_write(R_CTRL_1, 1)
		    board.digital_write(R_CTRL_2, 0)
		    board.analog_write(PWM_R, 245)
		    board.digital_write(L_CTRL_1, 1)
		    board.digital_write(L_CTRL_2, 0)
		    board.analog_write(PWM_L, 245)
		    board.sleep(2.0)

		    print("Left!")
		    board.digital_write(BOARD_LED, 0)
		    board.digital_write(R_CTRL_1, 0)
		    board.digital_write(R_CTRL_2, 1)
		    board.analog_write(PWM_R, 245)
		    board.digital_write(L_CTRL_1, 1)
		    board.digital_write(L_CTRL_2, 0)
		    board.analog_write(PWM_L, 245)
		    board.sleep(2.0)

		    print("Right")
		    board.digital_write(BOARD_LED, 1)
		    board.digital_write(R_CTRL_1, 1)
		    board.digital_write(R_CTRL_2, 0)
		    board.analog_write(PWM_R, 245)
		    board.digital_write(L_CTRL_1, 0)
		    board.digital_write(L_CTRL_2, 1)
		    board.analog_write(PWM_L, 245)
		    board.sleep(2.0)

		    print("Stop")
		    board.digital_write(BOARD_LED, 0)
		    board.digital_write(R_CTRL_1, 1)
		    board.digital_write(R_CTRL_2, 0)
		    board.analog_write(PWM_R, 0)
		    board.digital_write(L_CTRL_1, 1)
		    board.digital_write(L_CTRL_2, 0)
		    board.analog_write(PWM_L, 0)
		    board.sleep(5.0)

		if __name__ == "__main__":
		  setup()
		  while True:
		    loop()

closed 2015-08-14 by fisherds