MrYsLab/pymata-aio #20: Dropping commands after console input

fisherds, 2015-08-20:
We were trying to make a music example, but ran into more issues with dropped messages.  In the code below after getting a character from the user (using either input or msvcrt.getwch()) we need to do 2 things to get a message to send to the board.  We need to wait like 100ms and we have to send the command twice.  If we send it once it gets killed before the command gets to the board.  If we don't add a sleep then nothing ever sends to the board.  Any thoughts on where the messages die.  This is the same issue we had with the time.sleep() commands.  I'm sure it's a complex event loop issue within asyncio, but with a little guidance I'm happy to help investigate.  Anywhere here is the example code we were trying to add.  The code below works totally fine, but I hate having to send a command twice and saying to students "that's just how you have to do it".  Is there another way to get input from the console and not wreck the ship?

Hopefully you'll like this example btw...

        #!/usr/bin/python
        """
        Example of playing notes from the keyboard
        """
        
        from pymata_aio.pymata3 import PyMata3
        from pymata_aio.constants import Constants
        
        board = PyMata3()
        
        BUZZER_PIN = 9
        NOTE_C4 = 262
        NOTE_D4 = 294
        NOTE_E4 = 330
        NOTE_F4 = 349
        NOTE_G4 = 392
        NOTE_A4 = 440
        NOTE_B4 = 494
        NOTE_C5 = 523
        NOTE_D5 = 587
        NOTE_E5 = 659
        NOTE_F5 = 698
        
        
        def setup():
            board.set_pin_mode(BUZZER_PIN, Constants.OUTPUT)  # configures the buzzerPin as an OUTPUT
        
        
        def loop():
            key = input("Enter a key in the range asdfghjkl;' to play a note (Space to stop, x to exit) ")
            for i in range(2):  # Send the command twice
                board.sleep(0.1)  # Add a delay before each send
                if key == "a":
                    board.play_tone(BUZZER_PIN, Constants.TONE_TONE, NOTE_C4, None)
                elif key == "s":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_D4, None)
                elif key == "d":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_E4, None)
                elif key == "f":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_F4, None)
                elif key == "g":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_G4, None)
                elif key == "h":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_A4, None)
                elif key == "j":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_B4, None)
                elif key == "k":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_C5, None)
                elif key == "l":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_D5, None)
                elif key == ";":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_E5, None)
                elif key == "'":
                    board.play_tone(BUZZER_PIN,Constants.TONE_TONE, NOTE_F5, None)
                elif key == " ":
                    board.play_tone(BUZZER_PIN,Constants.TONE_NO_TONE, 0, 0)
                elif key == "x":
                    board.shutdown()
                    exit(0)
                else:
                    print("Unknown key")
                    break
        
        if __name__ == "__main__":
            setup()
            while True:
                loop()

MrYsLab, 2015-08-20:
@fisherds This was definitively a bug - thanks for finding it. Could you please try replacing the play_tone method in pymata3.py with the code [here](https://gist.github.com/MrYsLab/cdfabf59a9a500deb4f9) and let me know if that fixes it for you?

Also, you can remove your setup function, since the tone library will set the pin mode for you.

MrYsLab, 2015-08-20:
fixed with version 1.7

closed 2015-08-20 by MrYsLab

fisherds, 2015-08-20:
Confirmed that 1.7 fixed that issue.  Thanks again.