MrYsLab/pymata-aio #23: Teensy 3.1 errors out with unhelpful error message (suggestion)

mnapolitano89, 2015-09-16:
Teensy board throws the error "Initializing Arduino - Please wait... Cannot instantiate serial interface: /dev/tty.usbmodem958211", which is fixed by increasing arduino_wait - this isn't obvious to the end user as the needed step, suggesting that might be helpful :)

MrYsLab, 2015-09-16:
@mnapolitano89 How long of a delay does it need? I will put that into the documentation.

mnapolitano89, 2015-09-16:
changing to 5 worked for me

MrYsLab, 2015-09-16:
Thanks. I will document this. I am coming out with version 2 in the next few weeks, that is based upon Python 3.5 and I will make sure this gets incorporated into the documentation. Thanks again for letting me know.

mnapolitano89, 2015-09-16:
Thank you for writing this awesome bit of code! It's making my scientific life significantly easier

closed 2015-09-16 by mnapolitano89

MrYsLab, 2015-09-17:
@mnapolitano89 Wiki  example pages updated with a table of some typical arduino_wait times. Thanks for the kind words and great suggestion.