MrYsLab/pymata-aio #24: Error message when execute

euchcat, 2015-09-29:
When i execute 'pymata_aio_pymata3_example' with python 3.4, i receive this message from compiler:
Traceback (most recent call last):
  File "pymata3_test1.py", line 4, in <module>
    from pymata_aio.pymata3 import PyMata3
  File "/usr/local/lib/python3.4/dist-packages/pymata_aio/pymata3.py", line 23, in <module>
    from .pymata_core import PymataCore
File "/usr/local/lib/python3.4/dist-packages/pymata_aio/pymata_core.py", line 285
    async def start_aio(self):
            ^
SyntaxError: invalid syntax
why this message from python3.4? the syntax seem correct, any suggestion ?
Tanks for your job and sinceres greatings.
Lionel

MrYsLab, 2015-09-29:
For the latest version of pymata-aio, you need to run Python 3.5. The async/await syntax was introduced with that release. If you would still like to use Python 3.4.3, you can dowload the code from this branch https://github.com/MrYsLab/pymata-aio/tree/version_1.x.

I will close this issue. If you need any additional help on this issue please let me know. If you discover any other issues, please generate a new issue.

Thanks.

closed 2015-09-29 by MrYsLab

euchcat, 2015-09-30:
Hello,
I'd like to understand how 'pymata-aio' (version 2.5) has made for running
an installation in my distribution Python 3.4 with no error message.
I work on a linux machine and have only launched 'sudo pip3 install
pymata-aio' in command line. I received no warning from 'pip' during
installation and during the launch examples from python 3.4. (except the
error message than i have send you yesterday)
how do i make for installing pymata version 1.xx with pip3?
(I have already uninstall pymata 2.1 with pip3)
thanx for your response and sincere greetings.
Lionel

MrYsLab, 2015-09-30:
@euchcat To load 1.9, type the following command:
`pip3 install pymata_aio==1.9`

Please let me know if you still are having problems.

euchcat, 2015-10-01:
Hi Alan,
All is ok for the installation of pymata V1.x with PIP3.
I remark one thing when i execute the '3_sample.py' example:

pymata_aio Version 1.9    Copyright (c) 2015 Alan Yorinks All rights
reserved.
Using COM Port:/dev/ttyUSB0

Initializing Arduino - Please wait... Auto-discovery complete. Found 20
Digital Pins and 6 Analog Pins

Shutting down ...

------------------
(program exited with code: 1)
Press return to continue.

There is no problem with the code execution, why this message when program
exit? (after 'board.shutdown()')
Thanks for your help, it's just for understand how working pymata.
sincere greetings.

MrYsLab, 2015-10-01:
@euchcat I am glad you got things working. Shutdown tries to stop all of the running asyncio tasks, but it is not always successful. So upon  exiting your program, the event loop may throw an exception, which is what you are seeing. This should not cause any problems, except for the message you are seeing. Your program should be totally closed.

euchcat, 2015-10-01:
yes i have seeing that this message was just like a warning. i try to
create some GUI for the examples that you have on github.
I join you a python file if you want test him with pymata 1.9, this code is
working on my linux machine. If not there is no problem.
sinceres greetings.
Lionel