MrYsLab/pymata-aio #25: suggestion for board initialization output

callmeskywalker, 2015-10-04:
Hi, everyone. I want to say that I think this is a great tool. I am very happy at how easy it is to program my Arduino with Python. Believe it or not, my daughter actually thinks it is pretty cool too (whereas normally she is just like "Ehh, python.") I have one comment which is a matter of opinion, and perhaps there is a good reason why the developers choose to do this. I would be glad to hear the benefit/reasons for doing this. 

So I used the library for the first time today and I think it's really cool. However I saw this upon initializing the board. Again, this is a matter of opinion and in the grand scheme probably a small issue. Stylistically speaking I do not like the output of initializing the board:

    >>> board = PyMata3()

    pymata_aio Version 2.1    Copyright (c) 2015 Alan Yorinks All rights reserved.

    Using COM Port:/dev/ttyACM0

    Initializing Arduino - Please wait... Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

What I do not like about the output is the *mention of the copyright*. I feel that it takes away screen-space and distracts from the content relevant to programming. Personally I would just rely on the license in the project directory. I am basing this off of my use of the *scipy* stack. All of the packages I have used, do not mention copyright in the terminal. From what I can tell this is using GNU license, which I am sure at least some of the *scipy* stack uses. 

Anyways, I am huge fan so far! Looking forward making more use of the tool! I look forward to hearing back on this issue.

MrYsLab, 2015-10-04:
Glad you are enjoying the program. You can redirect the library output to a log file by :

board = PyMata3(log_output = True)

All other output will still go to the console.

Let me know if this gives you what you are looking for.

callmeskywalker, 2015-10-04:
Yeah that works but personally I think it would be better silent by default with the `board=PyMata3()` command. Maybe in addition to `log_output=True` there could be a `verbose=True`. The verbose option could show the current output from above.

MrYsLab, 2015-10-04:
@callmeskywalker One of the main reasons for the on-screen startup sequence, is that often, folks forget that they overwrote Firmata on their Arduinos with a sketch that pymata can't communicate with. The messages help debug those situations. In addition, by forcing current users to explicitly specify verbose mode at this point in time, would mean that those users would have to modify existing working code to see the messages. Lastly, many users do not read the changelog when updating their version, which would mean unnecessary support questions for me.

So, although your request is a legitimate one, I am sorry, but I will not be implementing it.

That being said, I appreciate the feedback and welcome any other questions or suggestions at any time.

closed 2015-10-04 by MrYsLab