MrYsLab/pymata-aio #27: V2.2 - v2.4 broke teensy3.x support

mnapolitano89, 2015-10-05:
Trying to figure out how or why it fails with the following:
Using COM Port:/dev/tty.usbmodem1053091

Initializing Arduino - Please wait... *** Analog map retrieval timed out. ***

Do you have Arduino connectivity and do you have a Firmata sketch uploaded to the board?

This is resolved by: specifying com_port, which promptly causes the following exception:
Traceback (most recent call last):

  File "/Users/mnapolitano/PycharmProjects/michaels_random_py/test.py", line 5, in <module>
    board = PyMata3(arduino_wait=5, com_port='/dev/cu.usbmodem1053091')
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pymata_aio/pymata3.py", line 56, in __init__
    com_port, ip_address, ip_port, ip_handshake)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pymata_aio/pymata_core.py", line 214, in __init__
    self.ip_address + ':' + str(self.ip_port))
TypeError: Can't convert 'NoneType' object to str implicitly
)

Commenting out ip support in pymata_core start() and setting the first if clause to always false allows for proper recognition

I believe the underlying issue to be here:
        else:
            if self.log_output:
                log_string = 'Using Ip Address/Port: ' + self.ip_address +\
                    ':' + str(ip_port)
                logging.info(log_string)
            else:
                print('Using Ip Address/Port: ' +
                      self.ip_address + ':' + str(self.ip_port))
should probably be:
        else:
            if self.log_output:
                log_string = 'Using Ip Address/Port: ' + self.ip_address +\
                    ':' + str(ip_port)
                logging.info(log_string)
            elif self.ip_address is not None:
                print('Using Ip Address/Port: ' +
                      self.ip_address + ':' + str(self.ip_port))

to stop that particular exception from messing things up

Why this is only the case for teensy and not for any of the others I don't know.

MrYsLab, 2015-10-05:
@mnapolitano89 
Thanks for letting me know about this.
I found the bug when inputting a fixed com port. At about line 207 (v2.4) of pymata_core.py,
I changed
```

        if self.com_port is None and self.ip_address is None:
            self.com_port = self._discover_port()
        else:
            if self.log_output:
                log_string = 'Using Ip Address/Port: ' + self.ip_address +\
                    ':' + str(ip_port)
                logging.info(log_string)
```

to

```

        if self.com_port is None and self.ip_address is None:
            self.com_port = self._discover_port()
        elif self.ip_address is not None:
            if self.log_output:
                log_string = 'Using Ip Address/Port: ' + self.ip_address +\
                    ':' + str(ip_port)
                logging.info(log_string)
```
I will see if I can add some code to not only look for tty.X but also cu.X automatically. I will let you know when I have something available. If you could test it for me (I don't own a Mac) that would be great.

Thanks.

MrYsLab, 2015-10-05:
@mnapolitano89 Looking through my old notes for [this issue](https://github.com/MrYsLab/pymata-aio/issues/5), I had modeled my auto discovery on what the node guys are doing. Not knowing if I add cu.XXXX into the mix will open up a whole new set of issues, I  prefer to leave auto discovery as is, and just fix the bug as shown above. What are your thoughts?

mnapolitano89, 2015-10-06:
I think that's a good idea, no sense adding more problems to the mix. The workaround is fairly simple as long as you know that's the problem

MrYsLab, 2015-10-06:
I will add a note on the Wiki page. The fix will be in the next release.

MrYsLab, 2015-10-11:
Fixed with release 2.5

closed 2015-10-11 by MrYsLab