MrYsLab/pymata-aio #29: Analog reporting issues?

mnapolitano89, 2015-10-06:
I've been running into some issues with analog polling in aio that I don't see in pymata plain, namely delayed polling and strange (inaccurate?) values. Looks something like this:
0
0
0
0
0
0
0
0
392
0
392
1007
392
1007
392
1007
566
1007

(I'm reading off of two pins, the first is pulled down to 0V, the second is pulled up to 3.3V) 

I'm getting similar strange readings on my mega, boarduino, trinket pro, and teensies so it isn't board related. I think I'm doing something dumb but not sure what. Program is:
import time
from pymata_aio_v9.pymata3 import PyMata3
from pymata_aio.constants import Constants

board = PyMata3(arduino_wait=10)
board.set_pin_mode(1, Constants.ANALOG)
board.set_pin_mode(2, Constants.ANALOG)

while(True):
    print(board.analog_read(1))
    print(board.analog_read(2))

    time.sleep(10)

fisherds, 2015-10-06:
Can you try switching from time.sleep(10) to board.sleep(10) for me to see if that helps.  In general asyncio and time.sleep don't play nicely together.  board.sleep() fixes that issue.  Just a thought

MrYsLab, 2015-10-06:
@mnapolitano89 Dave is correct. I did place a note about this on the [pymata3 usage wiki page.](https://github.com/MrYsLab/pymata-aio/wiki/Pymata3-Usage-Notes), but I know there is an awful lot of documentation and this might be easy to miss.

mnapolitano89, 2015-10-06:
I'll call it that, still having some issues but it might be hardware related...

closed 2015-10-06 by mnapolitano89