MrYsLab/pymata-aio #34: incompatible with pyserial==3.0

lukeorland, 2016-01-02:
I have to pin the version of pyserial to 2.7. Otherwise, this error occurs:

```
Initializing Arduino - Please wait... Traceback (most recent call last):
  File "example.py", line 9, in <module>
    board = PyMata3()
  File "/Users/lukeorland/.virtualenvs/electron-sprints/lib/python3.5/site-packages/pymata_aio/pymata3.py", line 59, in __init__
    self.core.start()
  File "/Users/lukeorland/.virtualenvs/electron-sprints/lib/python3.5/site-packages/pymata_aio/pymata_core.py", line 268, in start
    self.log_output)
  File "/Users/lukeorland/.virtualenvs/electron-sprints/lib/python3.5/site-packages/pymata_aio/pymata_serial.py", line 50, in __init__
    writeTimeout=1)
TypeError: __init__() got an unexpected keyword argument 'writeTimeout'
```

pymata-aio version 2.8

lukeorland, 2016-01-02:
related: https://github.com/hgrecco/pyvisa-py/issues/75#issuecomment-167890189

MrYsLab, 2016-01-02:
@lukeorland Thanks for letting me know about this. The pyserial author changed writeTimout to write_timeout, which makes pyserial 2.7 incompatible with 3.0. I have submitted an issue against pyserial 3.0 (https://github.com/pyserial/pyserial/issues/55).

I will make a note on my wiki page so that others can benefit by your discovery. 

If the pyserial author does not wish to fix this issue, I will then change my setup.py to restrict the library to 2.7.

MrYsLab, 2016-01-03:
@lukeorland FYI, pyserial was updated, but it did not fix the problem. I reopened this since it is clearly not fixed. I updated https://github.com/pyserial/pyserial/issues/55 so I will wait to see if there is a response.

MrYsLab, 2016-01-04:
@lukeorland I tested the latest changes to pyserial in github doing a local pip install and it appears that the problem is solved. There has not yet been a new release of pyserial to pypi so I will leave this open until the release becomes available and then I will verify to make sure it works.

MrYsLab, 2016-02-09:
Latest pyserial has solved the problem.

closed 2016-02-09 by MrYsLab