MrYsLab/pymata-aio #36: incorrect order of values in sonar_pin_entry

thiezn, 2016-02-09:
The documentation describes that the sonar_pin_entries in active_sonar_map is [cb, cb_type, value]. This is reflected properly in the sonar_config() function.

The pymata3-core sonar_data_retrieve() function however assumes the sensor value is in sonar_pin_entry[1] instead of sonar_pin_entry[2]. The same issue occurs in the  _sonar_data() function on lines 1568/1569.

MrYsLab, 2016-02-09:
@thiezn Another good catch. I had thought I had tested this feature, but it is apparent I had not. Thanks Again. This will also be in the updated release (2.9)

closed 2016-02-09 by MrYsLab

thiezn, 2016-02-09:
@MrYsLab thanks for the quick reply and fixes! I'll stop now whilst I'm ahead and go bragging to my friends I made my first (small) contribution to an open-source project! :+1:

thiezn, 2016-02-10:
@MrYsLab I noticed there's still a similar issue on lines 1568/1569 in the core library. They check if the value has changed but using sonar_pin_entry[1] instead of [2]. I don't think it breaks the code but does make unneccesary callbacks because of it.

MrYsLab, 2016-02-11:
Hmmm, I am seeing some other odd things in relation to Sonar, so I will be spending the day testing and will post an update when I am satisfied everything is working as it should.

MrYsLab, 2016-02-11:
@thiezn I am releasing version 2.10 which I believe resolves all of the sonar issues.

thiezn, 2016-02-11:
@MrYsLab works beautifully, thanks again!

MrYsLab, 2016-02-11:
@thiezn Thank you for letting me know and testing it out.