MrYsLab/pymata-aio #38: Problem with TKinter

HerpVanDerp, 2016-03-19:
Hello,

I've a problem with tkinter and pymata-aio:
I have a simpy window with two buttons. On one button i can switch a LED on and off, while the other detects a pressed button connected to the arduino and makes a windows with "pressed".
However if I press the arudino button, tkinter only creates the window "pressed", if I click on the LED button. I've noticed that there was a similar problem last year, but I'm not sure if it's a tkiner or pymata problem.

`from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants
from tkinter import *
from tkinter import ttk
from tkinter import messagebox
import time

def on():
    board.digital_write(BOARD_LED, 1)
    button.configure(bg = '#00CC00', text = "Off", command=off)
    pressed.configure(text = "Pressed")

    
def off():
    board.digital_write(BOARD_LED, 0)
    button.configure(bg = '#FF0101', text = "On", command=on)

def press(data):
    messagebox.showinfo(message='Button pressed')

    
def latch_callback(data):
        print(str(data))
        print(time.ctime(int(data[2])))
        
value = {}
BOARD_LED = 9
BOARD_BUTTON = 8
board = PyMata3(com_port='COM7')

root = Tk()
root.title('LED On/Off')
mainframe = ttk.Frame(root, padding="2 2 12 12")
mainframe.grid(column=0, row=0, sticky=(N, W, E, S))
mainframe.columnconfigure(0, weight=1)
mainframe.rowconfigure(0, weight=1)
button = Button(mainframe, text="On", command=on, background='red', width="30")
button.grid(column=2, row=1, sticky=W)
pressed = Button(mainframe, text="Not pressed", width = "30")
pressed.grid(column=2, row=2, sticky=W)

for child in mainframe.winfo_children(): child.grid_configure(padx=5, pady=5)
mainframe.focus()

def setup():
    board.set_pin_mode(BOARD_LED, Constants.OUTPUT)
    board.set_pin_mode(BOARD_BUTTON, Constants.INPUT, press)

if __name__ == "__main__":
    setup()
    root.mainloop()
`

MrYsLab, 2016-03-20:
The problem is that there are 2 event loops running, the tkinter loop, which is not asyncio compatible, and the asyncio loop and both are fighting each other. I don't think it is possible to run an asyncio program within a tkinter thread.

However, I tried rewriting your example for [pymata](https://github.com/MrYsLab/PyMata) and it works well. I adapted it for my Linux configuration and the pin numbers I am using. Just change those for your setup.

I am closing this issue. If you feel it should be reopened, please let me know.

```
from PyMata.pymata import PyMata
from tkinter import *
from tkinter import ttk
from tkinter import messagebox
import time

def on():
    board.digital_write(BOARD_LED, 1)
    button.configure(bg = '#00CC00', text = "Off", command=off)
    pressed.configure(text = "Pressed")

def off():
    board.digital_write(BOARD_LED, 0)
    button.configure(bg = '#FF0101', text = "On", command=on)

def cb_push_button(data):
    if data[2]:
        messagebox.showinfo(message='Button pressed')

value = {}
BOARD_LED = 9
BOARD_BUTTON = 12
board = PyMata("/dev/ttyACM0", verbose=True)

root = Tk()
root.title('LED On/Off')
mainframe = ttk.Frame(root, padding="2 2 12 12")
mainframe.grid(column=0, row=0, sticky=(N, W, E, S))
mainframe.columnconfigure(0, weight=1)
mainframe.rowconfigure(0, weight=1)
button = Button(mainframe, text="On", command=on, background='red', width="30")
button.grid(column=2, row=1, sticky=W)
pressed = Button(mainframe, text="Not pressed", width = "30")
pressed.grid(column=2, row=2, sticky=W)

for child in mainframe.winfo_children(): child.grid_configure(padx=5, pady=5)
mainframe.focus()

def setup():
    board.set_pin_mode(BOARD_LED, board.OUTPUT, board.DIGITAL)

    # Set the pin to digital input to receive button presses
    board.set_pin_mode(BOARD_BUTTON, board.INPUT, board.DIGITAL, cb_push_button)

if __name__ == "__main__":
    setup()
    root.mainloop()

```

closed 2016-03-20 by MrYsLab

fisherds, 2016-03-20:
This is a discussion that is overdue from us at Rose too.  For our usage of pymata-aio we really need to figure out how to use pymata-aio with Tkinter too.  We can't move to Pymata so that's not an option for us (technically we could move away from Tkinter, but not our first choice).  Without a doubt Tkinter with pymata-aio is our top priority to resolve (so I had to hop in on this discussion).  We were hoping to figure out solutions then start the discussion, but it looks like the discussion started before we were ready. :)  @HerpVanDerp here was a copy/paste of an email I got from my colleague Dave Mutchler about things he's tried.  pymata-aio and Tkinter CAN hackily happily work together sometimes sorta mostly already for at least a while.  I'll see if I can get Dave to comment on this thread as we are happy to put time into researching solutions.

Quote from Dave M:

> One workaround appears to be to have the Tk object (root) do a   root.after(small-time-1, f)   where  function f does nothing except sleep for a small-time-2 (thus returning control to asyncio’s loop) and then recurse via another root.after on itself.
>  
> I have confirmed that this workaround really does let both loops (Tk’s and asyncio’s) get some time with a short period.
>  
> Preliminary testing suggests that choosing the small time-1 and small-time-2 carefully lets asyncio’s loop “keep up” yet keep the GUI responsive.

HerpVanDerp, 2016-03-20:
Thank you very much for the help.
@fisherds approach works well.
In the main loop I added
```
if __name__ == "__main__":
    setup()
    root.after(10, sleep_2)
    root.mainloop()
```

and

```
def sleep_2():
        board.set_sampling_interval(10)
        root.after(10, sleep_2)

```

MrYsLab, 2016-03-20:
@HerpVanDerp Thanks for letting me know that it worked for you, and @fisherds thanks for the suggestion.

There is another potential solution that is on the asyncio side, namely using BaseEventLoop.run_in_executor and running either ProcessPoolExecutor or ThreadPoolExecutor.

I have concerns about both solutions. For the root.after solution, if one has a very complicated Tkinter GUI and there is a ton of data flying at the GUI, is it possible for some update to be missed? Not knowing the internals of Tkinter I would think that if the data changes are coming faster than the timeout period that this would be possible.

With the second solution - incorporating run_in_executor into PyMata3 may prove to be difficult and create a whole slew of other issues. 

So I am proposing a third solution - using 2 processes communicating with each other using ZeroMQ with a custom MessagePack protocol. @fisherds - any interest in something like this? What if I adapt rb4s to us a Tkinter GUI instead of Scratch. Initially I would provide motor control, have the ability to display the state of all of the sensors, and be able to reset the encoder counts.

I am not sure when I would get to this, but it would be an interesting extension to the Xideco concept.

fisherds, 2016-03-21:
Always thinking outside of the box Alan.  My first thought is that ZeroMQ with two processes sounds like a lot of trouble for a product that is 98% perfect. :)  I do always like new tools and additional options though.  We could use the existing tool for all of our non-GUI stuff then teach a 2 process communication tool when we get into GUIs.  So I'm interested, but it's probably a lower priority than just trying to find a solution for the existing tool.  It feels so close to working already that I hate to go back to the drawing board.  If you think it'd be fun to make the new 2 process tool though I'm happy to learn it. :)  I'll try to do some reading about BaseEventLoop.run_in_executor and we'll do some pushing on the root.after solution to see what limitations it has.  Dave M feel free to add any thoughts on the topic.

MrYsLab, 2016-03-21:
@HerpVanDerp 
I took another look and here is an alternative solution using PyMata3. Instead of having both event loops running, only the asyncio loop is being used and instead of calling root.after and root.mainloop, root.update is called within a loop owned by the application (see the end of the file).

```
import time                                                                                  
from tkinter import *                                                                        
from tkinter import messagebox                                                               
from tkinter import ttk                                                                      
                                                                                             
from pymata_aio.constants import Constants                                                   
from pymata_aio.pymata3 import PyMata3                                                       
                                                                                             
                                                                                             
def on():                                                                                    
    board.digital_write(BOARD_LED, 1)                                                        
    button.configure(bg='#00CC00', text="Off", command=off)                                  
    pressed.configure(text="Pressed")                                                        
                                                                                             
                                                                                             
def off():                                                                                   
    board.digital_write(BOARD_LED, 0)                                                        
    button.configure(bg='#FF0101', text="On", command=on)                                    
                                                                                             
                                                                                             
def press(data):                                                                             
    messagebox.showinfo(message='Button pressed')                                            
                                                                                             
                                                                                             
def latch_callback(data):                                                                    
    print(str(data))                                                                         
    print(time.ctime(int(data[2])))                                                          
                                                                                             
                                                                                             
value = {}                                                                                   
BOARD_LED = 9                                                                                
BOARD_BUTTON = 8                                                                             
board = PyMata3(com_port='COM7')                                                             
                                                                                             
root = Tk()                                                                                  
root.title('LED On/Off')                                                                     
mainframe = ttk.Frame(root, padding="2 2 12 12")                                             
mainframe.grid(column=0, row=0, sticky=(N, W, E, S))                                         
mainframe.columnconfigure(0, weight=1)                                                       
mainframe.rowconfigure(0, weight=1)                                                          
button = Button(mainframe, text="On", command=on, background='red', width="30")              
button.grid(column=2, row=1, sticky=W)                                                       
pressed = Button(mainframe, text="Not pressed", width="30")                                  
pressed.grid(column=2, row=2, sticky=W)                                                      
                                                                                             
for child in mainframe.winfo_children():                                                     
    child.grid_configure(padx=5, pady=5)                                                     
                                                                                             
mainframe.focus()                                                                            
                                                                                             
                                                                                             
def setup():                                                                                 
    board.set_pin_mode(BOARD_LED, Constants.OUTPUT)                                          
    board.set_pin_mode(BOARD_BUTTON, Constants.INPUT, press)                                 
                                                                                             
[... cut ...]
        try:                                                                                 
            root.update()                                                                    
            board.sleep(.0001)                                                               
        except KeyboardInterrupt:                                                            
            sys.exit(0)                                                                      
                                                                                             
```