MrYsLab/pymata-aio #41: Unpin pyserial==2.7

lfdebrux, 2016-08-11:
PyMata-aio was pinned to use pyserial==2.7 due to issue #34 , however that has since been fixed.

Would it be possible to now remove the version pinning from setup.py, as pyserial 2.7 is not compatible with python 3.5+

MrYsLab, 2016-08-11:
@lfdebru Could you please a sample of code or an example to explain what you mean by incompatible with Python 3.5?

lfdebrux, 2016-08-11:
This is the output I get on Windows 7 when trying to install pyserial 2.7

```
Processing dependencies for pymata-aio==2.15
Searching for pyserial==2.7
Reading https://pypi.python.org/simple/pyserial/
Best match: pyserial 2.7
Downloading https://pypi.python.org/packages/47/c9/7802e11ab388ad1539de716649add8bb8ca8bdff660364b3a404f79c27b7/pyserial
-2.7.win32.exe#md5=21555387937eeb79126cde25abee4b35
Processing pyserial-2.7.win32.exe
  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\rfc2217.py",
line 385
    except Exception, msg:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\serialcli.py"
, line 41
    except Exception, msg:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\serialjava.py
", line 69
    except Exception, msg:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\serialposix.p
y", line 58
    except IOError, e:
                  ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\serialutil.py
", line 535
    except TypeError, err:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\win32.py", li
ne 196
    MAXDWORD = 4294967295L # Variable c_uint
                         ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\tools\list_po
rts.py", line 83
    print "Filtered list with regexp: %r" % (args[0],)
                                        ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\tools\list_po
rts_linux.py", line 143
    print "%s: %s [%s]" % (port, desc, hwid)
                      ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\tools\list_po
rts_osx.py", line 207
    print "%s: %s [%s]" % (port, desc, hwid)
                      ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\tools\list_po
rts_posix.py", line 101
    print "%s: %s [%s]" % (port, desc, hwid)
                      ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\tools\list_po
rts_windows.py", line 240
    print "%s: %s [%s]" % (port, desc, hwid)
                      ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\tools\miniter
m.py", line 274
    except serial.SerialException, e:
                                 ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\urlhandler\pr
otocol_hwgrep.py", line 44
    print s
          ^
SyntaxError: Missing parentheses in call to 'print'

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\urlhandler\pr
otocol_loop.py", line 103
    except ValueError, e:
                     ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\serial\urlhandler\pr
otocol_socket.py", line 53
    except Exception, msg:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\okj43201\AppData\Local\Temp\easy_install-7_mnp2p0\pyserial-2.7-py3.5-win32.egg.tmp\EGG-INFO\scripts\min
iterm.py", line 274
    except serial.SerialException, e:
                                 ^
SyntaxError: invalid syntax

```

MrYsLab, 2016-08-11:
@lfdebrux I am not sure what procedure or tool you are using to install. I am using Windows 10, but it should be no different on Windows 7. When I use pip to install I have not issues. Here is the output of my installation:

```
C:\Users\Alan>pip install pyserial==2.7
Collecting pyserial==2.7
  Using cached pyserial-2.7.tar.gz
Installing collected packages: pyserial
  Running setup.py install for pyserial ... done
Successfully installed pyserial-2.7
```

If you do the same, do you still get those errors?

MrYsLab, 2016-08-11:
One other thing. If you have both python 2 and python 3 installed on your computer, make sure you are using the correct version of pip. On my computer, pip for python 3 is located in:
`
C:\Users\Alan\AppData\Local\Programs\Python\Python35\Scripts\pip.exe
`

The Python 2 version of pip on my computer is contained in:
`
c:\Python27\Scripts>
`

lfdebrux, 2016-08-12:
I tried your suggestion and now it works. Sorry for the bother.

closed 2016-08-12 by lfdebrux

MrYsLab, 2016-08-12:
@lfdebrux No bother at all. I am glad that it is working for you.