MrYsLab/pymata-aio #42: APC220 compatibility

Kenuat, 2016-08-23:
Hi, I'm trying to use pymata with radio module apc220 and it doesn't work. Console just prints the string "Initializing Arduino - Please wait..." and nothing happens, LEDs on the board are dead and I don't even get timout message or anything like that.
I've checked that the scheme I'm using is correct by aploading simple echo sketch and verifing that I see echoed strings at the PC.
Also I've found the similar issue described here: https://github.com/firmata/arduino/issues/296
It was fixed by sending reset command to the Arduino, I've tried the same, but without any luck, think I was doing this wrong. Also I've tried to press little red button on the board itself.
Hope to get any help or clues at least.

MrYsLab, 2016-08-23:
@Kenuat What Firmata sketch are you loading on to the Arduino? What model Arduino are you using?

Kenuat, 2016-08-23:
@MrYsLab Ops, sorry, forgot to write details
It is Arduino Uno and demo Firmata sketch which comes with Arduino IDE

MrYsLab, 2016-08-23:
@Kenuat Ok, let's first make sure without the apc220 things are functioning. If you just connect your arduino with the USB, does pymata-aio start up correctly?

If the answer is no, please provide a screen dump from the terminal window.

If the answer is yes, the did you set the speed of the apc220 to 57600? That is the baud rate that pymata-aio and firmata expects.

I am not familiar with the apc220, but pymata-aio expects to be talking to a standard com port.  What port is the apc220 mapped to?

Also, if you have no luck, if you can provide a wiring diagram of how you are hooking things up, that might be helpful.

I am not sure if this will be helpful, but I found this in the arduino forum:
```
Just configure the modules and set baud rates appropriately.
Be aware though that it cannot connect directly to an existing serial port on the PC. 
The APC220 has a TTL interface. You need to use a serial to USB converter for 
connection to a PC USB port, or a MAX232 driver to connect to a serial port directly.

Remember that Tx and Rx are applicable to their own module only. 
This means you have to cross over between modules. Arduino Tx goes to APC Rx pin when Txing.

PC Rx is connected to APC Tx. 
```

Kenuat, 2016-08-23:
@MrYsLab thank you for trying to help me, I very appreciate this

> If you just connect your arduino with the USB and no apc220 connected, does pymata-aio start up correctly?
Yes, while using USB everything works perfectly

> then did you set the speed of the apc220 to 57600? 
Yes, it is 57600, also I've tried to decrease it to various settings (both in sketch and in radio modules settings)

> What port is the apc220 mapped to?
Usually COM3, I send this name in pymata object constructor

> if you can provide a wiring diagram of how you are hooking things up, that might be helpful.
I'll try to make one, however I really doubt this diagram will be helpful, since the radio seems to work OK: basic echo sketch works (without firmata), and firmata uploaded board successfully sends startup string during boot:
`щрyS*t*a*n*d*a*r*d*F*i*r*m*a*t*a*.*i*n*o`
I've managed to see it using Serial Port Utility

MrYsLab, 2016-08-23:
@Kenuat Sorry, but since I do not own an apc220, I cannot debug the problem easily. When you start pymata-aio do you get any output to the console at all? You should see the same banner that you see when starting without the apc220 attached. If you do not, you might have to put a python debugger on pymata-aio and step through to see where it is hanging.

MrYsLab, 2016-08-24:
@Kenuat Have you been able to successfully run something like the experiment described here: [http://fritzing.org/projects/wireless-arduino-with-apc220](http://fritzing.org/projects/wireless-arduino-with-apc220) 

If you have, how log does it take both modules to sync up before you can communicate between the 2 modules. pymata-aio by default waits for 2 seconds to allow for the Arduino to reset itself before starting. You can increase this timer by using the **arduino_wait** parameter when instantiating the interface.

MrYsLab, 2016-08-24:
@Kenuat If STRING_DATA events are received by pymata-aio, they will automatically be sent to the console, but of course you need connectivity first to see them.

MrYsLab, 2016-08-24:
@Kenuat pymata-aio version 2.16 has just been released. If communication fails after 2 seconds when starting the application, the application is exited with an error message sent to the console.

MrYsLab, 2016-08-24:
@Kenuat I am closing this issue because the problem is clearly not within pymata-aio. If you are still having issues and I can help in any way, please don't hesitate to ask.

closed 2016-08-24 by MrYsLab

Kenuat, 2016-08-25:
Thank you for the new version and yes, I agree with you that it look like it is not pymata issue. I'm going to recheck everything step by step to be 100% sure that the whole stack is functional before returning to pymata.