MrYsLab/pymata-aio #44: Error message when execute in Arduino IDE

rricardi, 2016-08-31:
Hello,

When loading firmataplus generates a message and after that the compilation is completed. When I go to connect with Scratch, nothing happens. Can you help me? Thank you

The following error message:

`/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:39:1: warning: "/*" within comment [-Wcomment]
 /*
 ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:48:29: warning: extra tokens at end of #include directive
 #include <ooPinChangeInt.h> https://code.google.com/p/arduino-pinchangeint/downloads/detail?name=pinchangeint-v2.19beta.zip
                             ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:49:25: warning: extra tokens at end of #include directive
 #include <AdaEncoder.h> https://code.google.com/p/adaencoder/downloads/detail?name=adaencoder-v0.7beta.zip
                         ^
In file included from /home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:48:0:
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h: In constructor 'PCintPort::PCintPort(int, volatile uint8_t&)':
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h:200:12: warning: 'PCintPort::firstPin' will be initialized after [-Wreorder]
  PCintPin* firstPin;
            ^
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h:196:19: warning:   'const uint8_t PCintPort::PCICRbit' [-Wreorder]
  const  uint8_t   PCICRbit;
                   ^
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h:164:2: warning:   when initialized here [-Wreorder]
  PCintPort(int index,volatile uint8_t& maskReg) :
  ^
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h: In member function 'void PCintPort::PCint()':
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h:381:10: warning: unused variable 'thisChangedPin' [-Wunused-variable]
  uint8_t thisChangedPin, changedPins;
          ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino: In function 'void sysexCallback(byte, byte, byte*)':
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:849:36: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
           printData("argc = ", argc) ;
                                    ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:767:23: warning: variable 'encoder' set but not used [-Wunused-but-set-variable]
     static AdaEncoder encoder = 
                       ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:545:8: warning: unused variable 'pin' [-Wunused-variable]
   byte pin ;// used for tone
        ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:546:7: warning: unused variable 'frequency' [-Wunused-variable]
   int frequency ;
       ^
/home/renato/sketchbook/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:547:7: warning: unused variable 'duration' [-Wunused-variable]
   int duration ;
       ^
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h: In function 'attachInterrupt.constprop':
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h:319:18: warning: 'tmp' may be used uninitialized in this function [-Wmaybe-uninitialized]
  else tmp->next=p;
                  ^
/home/renato/sketchbook/libraries/ooPinChangeInt/ooPinChangeInt.h:297:12: note: 'tmp' was declared here
  PCintPin* tmp;
            ^`

MrYsLab, 2016-08-31:
The compiler warnings can be safely ignored. At one time the Arduino IDE had warnings turned off as a default, but in recent releases they have turned them on. You can set the warning levels in File/Preferences.

But that does not solve your problem. Please try running this simple blink script below with Python 3 and let me know the results. You should see the Pin 13 LED on the Arduino blink on and off. Do you see the LED blink? If you do, please give me defailed information about what is not working with Scratch. What are the blocks you are using, etc.  Also,what Model Arduino are you using and which operating system do you use?

By the way, to do something simple like using Scratch to control a digital output port, you can try using StandardFirmata. FirmataPlus is only necessary for some of the feature not supported by StandardFirmata (tone, etc.).

```
"""
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc
"""

from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants

BOARD_LED = 13
board = PyMata3()

def setup():
    board.set_pin_mode(BOARD_LED, Constants.OUTPUT)

def loop():
    print("LED On")
    board.digital_write(BOARD_LED, 1)
    board.sleep(1.0)
    print("LED Off")
    board.digital_write(BOARD_LED, 0)
    board.sleep(1.0)

if __name__ == "__main__":
    setup()
    while True:
        loop()
```

rricardi, 2016-09-01:
Thank you for the reply.

I'm already using StandardFirmata to manipulate LEDs and servo motors. Using StandardFirmata everything works normally, without any problems. But I need to expand my project to use the HC-SR04 Sensor and really need the use of Firmata Plus. I'm using an Arduino Nano and Linux operating system.

rricardi, 2016-09-01:
I removed the console warnings and apparently seems to have loaded the FirmataPlus but to load Scratch, nothing happens.

I'm using this circuit for my project.

[image]

and I'm running this code, just to test the sensor. I'm sure the sensor is working because I tested with a code in the IDE arduino and it shows the distance normally.

[image]

but always, distance show 0.

[image]

MrYsLab, 2016-09-01:
if you have firmataplus loaded and run the blink script does it work?

MrYsLab, 2016-09-01:
If it doesn't work, could you please copy and paste the output shown on the console.

rricardi, 2016-09-01:
you are referring to the blink code you passed or what is the example Arduino interface?

rricardi, 2016-09-01:
the code IDE Arduino is working correct. What you gave me is a problem when running in "from pymata_aio.pymata3 import PyMata3" line, has the following message:

```
Arduino: 1.6.11 (Linux) Board: "Arduino Nano, ATmega328"

teste_blink: 10: error: 'from' does not name a type
  from pymata_aio.pymata3 import PyMata3
  ^
exit status 1
'From' does not name a type
```

I think it should be with any structure of the wrong folder, would it?

MrYsLab, 2016-09-01:
I apologize, but I am not sure how you are running the blink script. Here is what I would expect to happen:
1. Copy the blink script above to a file. A good name would be blink.py.
2. With either StandardFirmata or FirmataPlus already installed on your nano, plug your Nano into a USB port.
3. Start the script from a terminal window by typing: python3 blink.py

What result do you see in the terminal window?

i am not

MrYsLab, 2016-09-01:
Did you follow the installation instructions located on this page to connect Scratch with your Nano? https://github.com/MrYsLab/s2aio/wiki

rricardi, 2016-09-01:
Yes, I can run the blink in Scratch, I have also created in Scratch a code for the servo. I added a buzzer in my circuit only for testing and it emits the sound. In that case it would only work if the Firmata Plus was installed in my nano, correct? If yes, then what the problem would be in my sensor does not work?

MrYsLab, 2016-09-01:
So if I understand you, the only problem you are having is with the SR-04. Here is a simplified script that communicates with the SR-04 directly. Could you please run it and let me know if you get any output? I you do not see any output, do you have any other way to determine if the SR-04 is working?

```
from pymata_aio.pymata3 import PyMata3

# ping callback function
def cb_ping(data):
    print(str(data[1]) + ' centimeters')

# create a PyMata instance
board = PyMata3(2)

# configure pins forSONAR module
board.sonar_config(6, 6, cb_ping)

while True:
    board.sleep(.1)

board.shutdown()

```

rricardi, 2016-09-01:
I ran another code in the Arduino IDE and it worked, I can get the SR-04 distance. I dont have another sensor to test but I suppose as this code worked, he has no problem. But then I can test this with your code to see if it works well.

MrYsLab, 2016-09-01:
Yes, if it works with another sketch, then we know the device is working. You will need to have FirmataPlus installed on the Nano before running the script above.

MrYsLab, 2016-09-01:
If that python script reports distances, than try adding a small wait in your Scratch forever loop (.2 for example).

rricardi, 2016-09-03:
After run the code _python3 hc-sr04.py_ for the SR-04, following this message:

```
Traceback (most recent call last):
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata3.py", line 22, in <module>
    from pymata_core import PymataCore
ImportError: No module named 'pymata_core'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "hc-sr04.py", line 1, in <module>
    from pymata_aio.pymata3 import PyMata3
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata3.py", line 24, in <module>
    from .pymata_core import PymataCore
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata_core.py", line 352
    async def start_aio(self):
            ^
SyntaxError: invalid syntax
```

rricardi, 2016-09-03:
After run the code for the SR04 in file _hc-sr04.py_, following this message:

```
Traceback (most recent call last):
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata3.py", line 22, in <module>
    from pymata_core import PymataCore
ImportError: No module named 'pymata_core'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "hc-sr04.py", line 1, in <module>
    from pymata_aio.pymata3 import PyMata3
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata3.py", line 24, in <module>
    from .pymata_core import PymataCore
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata_core.py", line 352
    async def start_aio(self):
            ^
SyntaxError: invalid syntax

```

MrYsLab, 2016-09-03:
It looks like you might be running Python 2 and not Python 3. Are you running Python 3?

rricardi, 2016-09-03:
Yes, i run with Python3. Now I run with Python 2 and show this message:

```
Traceback (most recent call last):
  File "hc-sr04.py", line 1, in <module>
    from pymata_aio.pymata3 import PyMata3
  File "/home/renato/sketchbook/pymata-aio/pymata_aio/pymata3.py", line 19, in <module>
    import asyncio
ImportError: No module named asyncio
```

MrYsLab, 2016-09-03:
From a console window, if you type:
`sudo pip3 freeze | grep pymata
`
does it report pyamata-aio?

rricardi, 2016-09-03:
nothing is reported :(

rricardi, 2016-09-03:
But if I type:
`pip freeze | grep PyMata`

Show this:
`PyMata==2.12`

MrYsLab, 2016-09-03:
Ok, so this means you have PyMata installed but not pymata-aio. When you use Scratch are you using s2a_fm or s2aio?

rricardi, 2016-09-03:
I use s2a_fm

MrYsLab, 2016-09-03:
That explains the errors above. s2a_fm uses PyMata and not pymata-aio. I will move this issue later to pymata.

Can you try running this script with Python 2: 
https://github.com/MrYsLab/PyMata/blob/master/examples/ping/ping_config_and_read.py

and this one:

https://github.com/MrYsLab/PyMata/blob/master/examples/ping/callback_ping_config_and_read.py

MrYsLab, 2016-09-03:
You may need to change the pin numbers and com port to match your system.

rricardi, 2016-09-03:
I change the pin numbers and the port.

The first code returns this message:

```
Python Version 2.7.10 (default, Oct 14 2015, 16:09:02) 
[GCC 5.2.1 20151010]

PyMata version 2.12  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyUSB0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 22
Total Number of Analog Pins Detected = 8
Traceback (most recent call last):
  File "hc-sr04.py", line 46, in <module>
    print(str(data[2]) + ' centimeters')
KeyError: 2
```

And the second code, this:
```
Python Version 2.7.10 (default, Oct 14 2015, 16:09:02) 
[GCC 5.2.1 20151010]

PyMata version 2.12  Copyright(C) 2013-16 Alan Yorinks    All rights reserved.

Opening Arduino Serial port /dev/ttyUSB0 

Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board initialized in 0 seconds
Total Number of Pins Detected = 22
Total Number of Analog Pins Detected = 8
0 centimeters
PyMata close(): Calling sys.exit(0): Hope to see you soon!
```

MrYsLab, 2016-09-03:
I am not at my lab computer right now, so I will try running this tomorrow to see what might be happening. I will get back to you tomorrow.

rricardi, 2016-09-03:
Ok, i'm waiting :weary:

rricardi, 2016-09-03:
Do you agree with my scratch code that went before? It seems to be ok?

MrYsLab, 2016-09-03:
It looks like you are running things correctly. I will need to verify tomorrow. and I will let you know my results.

MrYsLab, 2016-09-03:
@rricardi There was an issue with PyMata and it has been corrected. If you run the following command:

`
sudo pip2 install PyMata --upgrade
`

You should get the latest version with the fix and Sonar should now work for you.

I moved this issue to PyMata issue #26 and therefore I am closing the issue here.

Please test and let me know if things are now working for  you.

MrYsLab, 2016-09-03:
Please leave your results on the PyMata issue https://github.com/MrYsLab/PyMata/issues/26

closed 2016-09-03 by MrYsLab