MrYsLab/pymata-aio #45: FirmataPlus Arduino IDE Compilation Error

masteredward, 2016-09-28:
Hello,

My Arduino IDE is reporting some errors, aborting the compilation. I'm using version 2:1.0.5 of Arduino IDE. Here is the error:

FirmataPlus.ino: In function ‘void setPinModeCallback(byte, int)’:
FirmataPlus.ino:339:20: error: ‘disableI2CPins’ was not declared in this scope
FirmataPlus.ino:347:77: error: ‘reportAnalogCallback’ was not declared in this scope
FirmataPlus.ino: In function ‘void sysexCallback(byte, byte, byte*)’:
FirmataPlus.ino:680:23: error: ‘enableI2CPins’ was not declared in this scope
FirmataPlus.ino:871:36: error: ‘printData’ was not declared in this scope

Can you give me some advice? Thanks!

MrYsLab, 2016-09-28:
The version of the IDE that the code is tested against is 1.6.12 and can be downloaded here: https://www.arduino.cc/en/Main/Software

Please try using this version and let me know if you are still having issues or if this solves the problem.

masteredward, 2016-09-28:
Problem solved! I was using the version distributed in Ubuntu Xenial Repo (1.0.5). Compiled only with warnings! Thanks!

closed 2016-09-28 by masteredward