MrYsLab/pymata-aio #47: Error after initializing with the board

adamdrenth, 2016-11-07:
I am having this issue on my windows computer.  I am running Windows 10 and PyCharm CE with the pymata-aio library. I also have the pymata library but it was not imported into the project.  I have FirmataPlus loaded on the Arduino. However, I cannot get any python program to run.  It will work on my Mac machine though... The error output from PyCharm is below.

C:\Users\tbaxt\AppData\Local\Programs\Python\Python35-32\python.exe C:/Users/tbaxt/Downloads/pymata-aio-master/pymata-aio-master/examples/blink.py
 
pymata_aio Version 2.16              Copyright (c) 2015-2016 Alan Yorinks All rights reserved.
 
Using COM Port:com6
 
Initializing Arduino - Please wait... Traceback (most recent call last):
  File "C:/Users/tbaxt/Downloads/pymata-aio-master/pymata-aio-master/examples/blink.py", line 15, in <module>
    board = PyMata3()
  File "C:\Users\tbaxt\AppData\Local\Programs\Python\Python35-32\lib\site-packages\pymata_aio\pymata3.py", line 62, in __init__
    self.core.start()
  File "C:\Users\tbaxt\AppData\Local\Programs\Python\Python35-32\lib\site-packages\pymata_aio\pymata_core.py", line 297, in start
    print("\nArduino Firmware ID: " + firmware_version)
TypeError: Can't convert 'NoneType' object to str implicitly
Task was destroyed but it is pending!
task: <Task pending coro=<PymataCore._command_dispatcher() running at C:\Users\tbaxt\AppData\Local\Programs\Python\Python35-32\lib\site-packages\pymata_aio\pymata_core.py:1301> wait_for=<Future finished result=None>>
 
Process finished with exit code 1

adamdrenth, 2016-11-07:
I should have included the program I ran which caused the error:

from pymata_aio.pymata3 import PyMata3      
from pymata_aio.constants import Constants  

# Declare any Global Variables
board = PyMata3()
LedPin = 13

# Initialize your pins
board.set_pin_mode(LedPin, Constants.OUTPUT)

# Do something
board.digital_write(LedPin, 1)

# End the Program.  
board.shutdown()

MrYsLab, 2016-11-07:
@adamdrenth Windows 10 should not be an issue, and I use PyCharm (on linux), so that too should not be an issue. So, some questions:

1. Are you running the same firmata sketch loaded on your Arduino for both the Mac and Windows tests?

2. In Windows, if you bring up the Arduino IDE, is COM6 the correct COM port? If you have something plugged into COM6, but your Arduino is actually at a higher number com port, pymata-aio can't detect this and it grabs the first com port it sees.

If you need to set the com port you would use: 
`
board = PyMata3(com_port=COM7)
`
assuming that COM7 is the one your Arduino is using - modify to meet your system.

If the com6 is correct, then try unplugging the Arduino cable, put it back in, wait about 5 seconds and then try running the program again.

Please let me know if any of this works or does not. If not, please let me know which type of Arduino you are using.

adamdrenth, 2016-11-08:
It didn't work unfortunately.  When I move from port to port the computer automatically assigns it the same number (com 3).  I tried to force com 3 as well, but no difference. I've tried changing the board and USB cable just make sure that is not the issue.  I have noticed that when I run my basic program, the lights flash as if there is some communication, but I'm still getting the same error.

MrYsLab, 2016-11-08:
@adamdrenth It is not clear how you are deciding which com port to use. If I open up my device manager, I can see what windows has chosen com 5 as shown below. If you look at your device manager, is this the com port that pymata has automatically chosen and/or the one you are specfiying?

[image]

adamdrenth, 2016-11-08:
I have tried both letting Pymata pick the port and setting it in my code.  Pymata is able to detect which port the Arduino is connected to and I am receiving the error.  I have noticed that my machine periodically changes the port names. It was 6 before and now its 3 which may have lead to some confusion.

MrYsLab, 2016-11-08:
@adamdrenth Using the Arduino IDE on your Windows computer, are you able to load StandardFirmata onto your Arduino?

adamdrenth, 2016-11-08:
I am able to load the standardfirmata and firmataplus onto the arduino. I have firmataplus loaded on the arduino board.

MrYsLab, 2016-11-08:
@adamdrenth Does changing 
`
board = PyMata3()
`

to

`
board = PyMata3(arduino_wait=5)
`

produce the same error?

MrYsLab, 2016-11-09:
@adamdrenth Are you still having issues or can I close this issue?

adamdrenth, 2016-11-09:
I didn’t have a chance to test it yesterday or today.  I will be able to get back you by tomorrow.

adamdrenth, 2016-11-10:
Hmm.  Still not working. I've noticed that its using COM3 now and it says that COM4 is in use.  I tried to override it and force COM4 to be the Arduino with no effect.  I also tried changing the baud rate to 57600 for the com port to no effect.  The default is 9600.

I will say I am using this library with my class and its working for a majority of them.  But my computer and two others are having the same issue.  All windows and all with the same symptoms. Mac's are so much more stable..

MrYsLab, 2016-11-10:
I found this [link](https://startingelectronics.org/software/arduino/installing-arduino-software-windows-10/) that talks about installing an Arduino USB seriial driver.

Windows 10 has been pretty stable, so it is either a driver issue or something in your environment. If you have no other serial devices connected other than the Arduino, I don't understand why the port keeps changing. That sometimes happens with a Leonardo but I have never seen it with an Uno. Are you using an "official" Arduino Uno? Some of the clones use different circuitry/firmware which may make the board behave like a Leonardo in regard to serial communication.

Let me know if any of this helps.

MrYsLab, 2016-11-10:
I am going to close this issue. If you have any other questions, you can add them here if you wish.

closed 2016-11-10 by MrYsLab

adamdrenth, 2016-11-18:
I wanted to say that the wait time fixed all of the issues. This issue only happened on Windows machines. The only other time I had an issue was when a board didn't have actually have the firmata installed.

MrYsLab, 2016-11-18:
Thanks for letting me know. I am glad that it worked for you.