MrYsLab/pymata-aio #48: Compiling Error "Linking everything together..." FirmataPlus

fischer1275, 2016-11-08:
Hello, 

I am getting multiple error messages while trying to compile the FirmataPlus library onto my Arduino Uno. I am using the Arduino IED 1.6.12 on a Win 7 Pro system. The console gives me out the error during the part where it tries to link everything together: 

`Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -w -Os -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "C:\Users\csc\AppData\Local\Temp\arduino_build_238763/FirmataPlus.ino.elf" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\sketch\FirmataPlus.ino.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\Servo\avr\Servo.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\Servo\sam\Servo.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\Servo\samd\Servo.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\Wire\Wire.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\Wire\utility\twi.c.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\FirmataPlus\FirmataPlus.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\NewPing\NewPing.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\Stepper\Stepper.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o" "C:\Users\csc\AppData\Local\Temp\arduino_build_238763/core\core.a" "-LC:\Users\csc\AppData\Local\Temp\arduino_build_238763" -lm
C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::cbmethod()'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::turnOffPWM(unsigned char)'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::getFirstEncoder()'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `firstEncoder'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::attachInterrupt(unsigned char, unsigned char)'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::addEncoder(char, unsigned char, unsigned char)'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::genie()'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `currentEncoder'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::getID()'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::getClicks()'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder.cpp.o (symbol from plugin): In function `AdaEncoder::cbmethod()':

(.text+0x0): multiple definition of `AdaEncoder::query()'

C:\Users\csc\AppData\Local\Temp\arduino_build_238763\libraries\AdaEncoder\AdaEncoder (2).cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Using library Servo at version 1.1.2 in folder: C:\Program Files (x86)\Arduino\libraries\Servo 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library FirmataPlus in folder: C:\Users\csc\Documents\Arduino\libraries\FirmataPlus (legacy)
Using library NewPing in folder: C:\Users\csc\Documents\Arduino\libraries\NewPing (legacy)
Using library Stepper at version 1.1.3 in folder: C:\Program Files (x86)\Arduino\libraries\Stepper 
Using library ooPinChangeInt in folder: C:\Users\csc\Documents\Arduino\libraries\ooPinChangeInt (legacy)
Using library AdaEncoder in folder: C:\Users\csc\Documents\Arduino\libraries\AdaEncoder (legacy)
exit status 1
Error compiling for board Arduino/Genuino Uno.
`

Hope you could help. 
Thanks!

MrYsLab, 2016-11-08:
@fischer1275 It looks as if you might have duplicate copies of the library files. Please look at your directory:

`
C:\Users\csc\Documents\Arduino\libraries
`

And make sure that there is not another libraries directory below that. This will look like this:

`
C:\Users\csc\Documents\Arduino\libraries\libraries
`

There should only be the top level libraries directory. The files contained in lower level must be moved up to the upper level and then the lower level removed.

Here is what my directory structure looks like:

```
 Directory of C:\Users\Alan\Documents\Arduino\libraries

07/06/2016  08:55 AM    <DIR>          .
07/06/2016  08:55 AM    <DIR>          ..
06/29/2016  11:04 AM    <DIR>          AdaEncoder
06/29/2016  11:04 AM    <DIR>          cbiface
06/29/2016  11:04 AM    <DIR>          cppfix
07/06/2016  08:55 AM    <DIR>          EnableInterrupt
07/06/2016  08:55 AM    <DIR>          Firmata
06/29/2016  11:04 AM    <DIR>          FirmataPlus
06/29/2016  11:04 AM    <DIR>          FirmataPlusLBT
06/29/2016  11:04 AM    <DIR>          FirmataPlusRB
06/29/2016  11:04 AM    <DIR>          FirmataPlusRBPixy
06/29/2016  11:04 AM    <DIR>          NewPing
06/29/2016  11:04 AM    <DIR>          ooPinChangeInt
06/29/2016  11:04 AM    <DIR>          Pixy
06/29/2016  11:04 AM                87 readme.txt
06/29/2016  11:04 AM    <DIR>          RedBot
06/29/2016  11:04 AM    <DIR>          RedBot2
07/06/2016  08:55 AM    <DIR>          Temboo
               1 File(s)             87 bytes
              17 Dir(s)  19,751,313,408 bytes free
```
I probably have a few extra things, but you can see the heirarchy.

Also, the default for the latest Arduino IDE has compiler warnings turned on. You may see a bunch of warnings after compiling. The warnings (not the errors if any) can be safely ignored.

Please let me know if you have solved the issue or not.

fischer1275, 2016-11-08:
@MrYsLab Thank you for your fast response. 

I have checked my libraries folder, but it looks exactely as yours (minus the Temboo folder). 

` Directory of C:\Users\csc\Documents\Arduino\libraries

08/11/2016  10:23    <DIR>          .
08/11/2016  10:23    <DIR>          ..
08/11/2016  10:23    <DIR>          AdaEncoder
08/11/2016  10:23    <DIR>          cbiface
08/11/2016  10:23    <DIR>          cppfix
06/07/2016  07:25    <DIR>          EnableInterrupt
06/07/2016  07:25    <DIR>          Firmata
08/11/2016  09:48    <DIR>          FirmataPlus
08/11/2016  09:48    <DIR>          FirmataPlus32u4
08/11/2016  10:23    <DIR>          FirmataPlusLBT
08/11/2016  10:23    <DIR>          FirmataPlusRB
08/11/2016  10:23    <DIR>          FirmataPlusRBPixy
08/11/2016  10:23    <DIR>          NewPing
08/11/2016  10:23    <DIR>          ooPinChangeInt
08/11/2016  10:23    <DIR>          Pixy
29/07/2016  08:22               176 readme.txt
08/11/2016  10:23    <DIR>          RedBot
08/11/2016  10:23    <DIR>          RedBot2
26/10/2016  10:10    <DIR>          Rotary
               1 File(s)            176 bytes
              18 Dir(s)  425,675,935,744 bytes free`

So I am at a loss what the issue seems to be. 

Best regards

MrYsLab, 2016-11-08:
@fischer1275 Temboo is something that the IDE added as part of an update and can be ignored.

Two questions:
1. When you are going to compile, are you selecting from File/Examples/FirmataPlus/FirmataPlus to load the sketch?

2. Do you have the Uno selected in Tools/Board?

fischer1275, 2016-11-08:
@MrYsLab 

Yes, I chose the FirmataPlus from Examples under "Examples from Custom libraries"/FirmataPlus/FirmataPlus. 

And I also select the ArduinoUno on the Tools/Board. 

I do not know if this is of any value, but I tried compiling other libraries of the package you provide. I also receive an error when trying to compile the MyEncoder from the AdaEncoder library. It gives me this Error: 

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\csc\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10612 -build-path C:\Users\csc\AppData\Local\Temp\arduino_build_890283 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\csc\Documents\Arduino\libraries\AdaEncoder\Examples\MyEncoder\MyEncoder.pde
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\csc\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10612 -build-path C:\Users\csc\AppData\Local\Temp\arduino_build_890283 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\csc\Documents\Arduino\libraries\AdaEncoder\Examples\MyEncoder\MyEncoder.pde
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Build options changed, rebuilding all
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\csc\AppData\Local\Temp\arduino_build_890283\sketch\MyEncoder.pde.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\csc\AppData\Local\Temp\arduino_build_890283\sketch\MyEncoder.pde.cpp" -o "C:\Users\csc\AppData\Local\Temp\arduino_build_890283\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\Users\csc\Documents\Arduino\libraries\AdaEncoder\Examples\MyEncoder\MyEncoder.pde:2:24: fatal error: ByteBuffer.h: No such file or directory

 #include <ByteBuffer.h>

                        ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.
```

Seeing that FirmataPlus is calling the AdaEncoder during compilation, could this be a cause of the error ? 

Thank you for your patience

MrYsLab, 2016-11-08:
@fischer1275 It can if you are using an Arduino with a32u4 processor. You can try compiling FirmataPlus32u4 which removes the adaencoder from the mix. All other functionality is present in that build.

Please let me know if that solves your problem.
Thanks.

fischer1275, 2016-11-08:
@MrYsLab  I can compile the FirmataPlus32u4 without any problems, however I suspect that the adaencoder is the part of the FirmataPlus which is needed to give it the functionality to control rotary encoders, correct?  Which would be exactly why I wanted to use the FirmataPlus as it combines the Firmata protocol with access to a rotary encoder function.

MrYsLab, 2016-11-08:
@fischer1275 It looks as if compilation if failing on the examples which are not needed for FirmataPlus. For some reason, it can't find ByteBuffer.h, which is in the examples directory.  I just removed the Examples directory from the AdaEncoder directory and was able to recompile without issue. I have no clue why you might be getting that error, but you might want to give this a try.

fischer1275, 2016-11-08:
@MrYsLab  I did that as well, unfortunately it did not resolve the issue with failed compilation of the FirmataPlus. The error as described above persists.

MrYsLab, 2016-11-08:
@fischer1275 Sorry, I was looking at your attempt at compiling the adafruit encoder library. Going back to your original set of errors, it is complaining about things being multiply defined. Do you have another copy of the AdaEncoder Library anywhere?

MrYsLab, 2016-11-08:
@fischer1275 Libraries are also stored in C:\Program Files(x86)\Arduino\libraries so please check that directory as well.

fischer1275, 2016-11-08:
@MrYsLab I found the issue of the compiling error. While extracting the zip files and copy the library section to the Arduino directory I made the mistake and merged the folders instead of replace them. This caused me to have 2 AdaEncoder.cpp files in that folder. I deleted one of them and now the compilation of FirmataPlus ran without any errors and I was able to copy it onto my ArduinoUno, 

Thank you very much for you help and time !

MrYsLab, 2016-11-08:
[PyMata](https://github.com/MrYsLab/PyMata) works with Python 2 and 3 (although I have seen some issues with 3). It uses threading and not asyncio, so it is not as efficient at pymata-aio. Pymata-aio is only Python 3.5 and above.

MrYsLab, 2016-11-08:
I will be closing the issue, but if you have any other questions you can still post them here.

closed 2016-11-08 by MrYsLab

fischer1275, 2016-11-09:
@MrYsLab  Thank you for all your help. I had a question regarding the rotary encoder function of the FirmataPlus. I saw a code you posted as answer to a different question: 

```
# encoder test script
from PyMata.pymata import PyMata

encoder_pin_a = 2
encoder_pin_b = 3

#  Create a PyMata instance using the Arduino COM Port as the input parameter
firmata = PyMata("COM5")

# Configure
firmata.set_pin_mode(encoder_pin_a, firmata.INPUT, firmata.ENCODER)
firmata.set_pin_mode(encoder_pin_b, firmata.INPUT, firmata.ENCODER)
firmata.encoder_config(encoder_pin_a, encoder_pin_b)

# The current value
while (1):
    print(firmata.digital_read(encoder_pin_a))
```
 I tried it out but the output is somewhat strange. I am using an incremental [rotary encoder](https://www.sparkfun.com/products/11102) but when I turn the rotary encoder I the values change between showing 0 and 1 and other values like -4200 incrementing at the lightest touch and then going to a positive value and switch back. I can see no logic behind the jumps in numbers. 

Best, 
Max

MrYsLab, 2016-11-09:
@fischer1275 The adaencoder library works  with 2 pin encoders as mentioned [here](https://code.google.com/archive/p/adaencoder/). The part is similar to t[his] (https://www.adafruit.com/products/377?gclid=CKPImKzJm9ACFdhMDQodrB8JrA) part.  Unfortunately it is not compatible with your encoder.