MrYsLab/pymata-aio #50: how to install on raspberry pi?

ash73, 2017-02-05:
Hi, I'm looking for a way to control an Arduino Uno from a Raspberry Pi using Python 3.

Firstly, is pymata-aio the right way to do this?

Secondly, how to get it working? I couldn't find any instructions so I tried the following:

git clone https://github.com/tino/pymata-aio
cd pymata-aio
sudo python setup.py build
sudo python setup.py install

It gave a ton of error messages.

Beginner instructions please.

MrYsLab, 2017-02-05:
If you are using Python 3.4.2 that comes with Raspbian, you will not be able to use pymata-aio since it requires Python 3.5 or greater. 

If you are using Python 3.5 or greater than to install you can use pip:

```
sudo pip3 install pymata-aio
```

Please let me know if you have any other questions.

ash73, 2017-02-05:
Thanks, that explains it.

What is the relation between pymata-aio and pymata? I'm trying to figure out which one to use.

MrYsLab, 2017-02-05:
PyMata handles concurrency using Python threads, while pymata-aio uses the Python asyncio module. The performance (cpu utilization) for pymata-aio is much better than PyMata. I wrote a sample program to control a servo motor for both libraries. The CPU utilization for PyMata was 34% and pymata-aio was 5%.

The pymata-aio library comes with 2 APIs. One is called pymata_core and its documentation can be found here:
http://htmlpreview.github.io/?https://github.com/MrYsLab/pymata-aio/blob/master/documentation/html/pymata_core.html

It requires one to have a knowledge of ayncio.

There is another API called pymata_3, which acts as a proxy for pymata_core, and requires a lot less knowledge of asyncio.  The API may be found here:

http://htmlpreview.github.io/?https://github.com/MrYsLab/pymata-aio/blob/master/documentation/html/pymata3.html

Both PyMata and pymata-aio work well, but pymata-aio has the better performance and is also the newer library. Both are being maintained, so if you have any issues with either, they will addressed.

BTW, I wrote an article, http://mryslab.blogspot.com/search/label/Esplora, 
that describes my finding using asyncio for two projects using the Arduino Esplora board. There are 2 graphs in the article that show the performance of a threading model versus using asyncio.

MrYsLab, 2017-02-06:
I am closing this issue. If you have any other questions, you can add them here.

closed 2017-02-06 by MrYsLab