MrYsLab/pymata-aio #51: how to invoke pymata_aio socket for web control?

cryoguy, 2017-02-10:
A question from a newbie:  I am getting "The socket has closed! Did you start the server?" when I try the example ( pymata-aio/examples/pwm_pin_example.html).   what do I do to invoke the socket?  

I am using Arch linux with newest version of everything as of now.

Thanks

Tony

MrYsLab, 2017-02-11:
Hi Tony,
Do you have a Firmata sketch loaded on your Arduino? If not, you can find instructions here:
https://github.com/MrYsLab/pymata-aio/wiki/Uploading-StandardFirmata-To-Arduino

Also, is your Arduino plugged into your computer?

cryoguy, 2017-02-11:
Thank you for your prompt reply.   Firmata is loaded and the pymata-core example works fine.   For some reason just the socket does not work.  I am using a Mega board.

MrYsLab, 2017-02-11:
You are indeed correct. I tried on multiple platforms and they all fail the same way. The problem is that the websocket connection is being refused. I will need to investigate, what is going on with websockets. Hopefully I will have an answer soon.

MrYsLab, 2017-02-11:
How embarrassing.  I forgot you need to start pymata_iot.py before bringing up the web pages. I just modified the wiki page to say that.
Please let me know if that solves the problem or not. Thanks.

cryoguy, 2017-02-11:
Thanks, that solves the problem.

closed 2017-02-12 by MrYsLab