MrYsLab/pymata-aio #52: ping.py with HC-SR04 not printing values

pragyhazarika, 2017-02-17:
Hi, I'm using FirmataPlus on an Arduino Uno and I am running the attached code on Windows 10, however I get the following message and nothing happens after that.

> pymata_aio Version 2.16 Copyright (c) 2015-2016 Alan Yorinks All rights reserved.
> 
> Initializing Arduino - Please wait...
> Arduino Firmware ID: 2.5 FirmataPlus.ino
> Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

I have connected the trigger and echo pins of the HC-SR04 sensor to pin 12 of the Uno and connected the 5V and GND as well. I tried connecting the trigger and echo pins to different Arduino pins but that didn't help. 

I would like to add that sometimes I do get the sonar data printed but they are wrong, at other times the data would be correct but would not update for a long time (say 10 seconds or so). Any advice on what I could try?

Here is the code I am running.    

`
from pymata_aio.pymata3 import PyMata3

# ping callback function
def cb_ping(data):
    print(str(data[1]) + ' centimeters')

#create a PyMata instance
#board = PyMata3(3)

board=PyMata3(arduino_wait=5, com_port='COM8')

# configure 4 pins for 4 SONAR modules
board.sonar_config(12, 12, cb_ping)

while True:
    board.sleep(.1)
board.shutdown()
`

MrYsLab, 2017-02-17:
I just ran the following script on windows 10, and it is working correctly for me. Do you have any way to verify that your hc-sr04 is functioning properly?

```
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants

board = PyMata3(arduino_wait=5, com_port="com6")

def cb_ping(data):
    print(data)
    
board.sonar_config(12, 12, cb_ping)

while True:
    board.sleep(.1)

```

pragyhazarika, 2017-02-17:
Hi,

I tested hc-sr04 with a Arduino sketch (NewPing example) and it works fine.

An update: My original code has now started printing values but the issue of lagged printing still persists. I also tested your code above and it presents the same issue.

MrYsLab, 2017-02-17:
The print statement will only occur for changes in values from the HC-SR04, so if there are no changes, then it will appear things are lagging. Could that be the problem you are seeing?

pragyhazarika, 2017-02-17:
I don't think that's the problem. For instance, if I point the HC-SR04 towards the wall (about 50cm away), the printed value is 50cm, however when I move my hand in between it takes several seconds for the printed value to be updated. Moreover the update process is quite not uniform and choppy. I'm not an expert on os but it seems as though Windows(or Python?) is unable to keep up with the data from the Arduino.   

Update: I ran my original code as well as your code above again. Now, neither prints anything after the auto-discovery step of the analog and digital pins.

MrYsLab, 2017-02-17:
Would you be willing to try the [PyMata](https://github.com/MrYsLab/PyMata) library? This will run with either Python 2 or 3 and does not use asyncio.

If want to try it, install the library with:

pip install PyMata

And then run [this script](https://github.com/MrYsLab/PyMata/blob/master/examples/ping/callback_ping_config_and_read.py)

It will run for 10 seconds and then exit. Please let me know if the results are better.

pragyhazarika, 2017-02-17:
Hi,

I used the PyMata library (I am running Python 3.6) and all issues with lag have been resolved. I imagine asyncio wasn't working very well for me. Thank you for your prompt help!

closed 2017-02-17 by pragyhazarika