MrYsLab/pymata-aio #53: Replace print statements with python built in loggin

ajd394, 2017-03-21:
Using print statements pollutes STDOUT when using pymata aio in a larger python program. By using python's build in logging, a developer can choose where the logging will appear more easily.

MrYsLab, 2017-03-21:
@ajd394 I am not understanding your comment. You can set output to go to log when invoking pymata_core or pymata3.

Here is the parameter list for __init__ in pymata_core:

```
def __init__(self, arduino_wait=2, sleep_tune=0.0001, log_output=False,
                 com_port=None, ip_address=None, ip_port=2000,
                 ip_handshake='*HELLO*'):
```

If you set log_output to True, print statements will be sent to pymata_aio.log.

Does this solve your problem?

ajd394, 2017-03-21:
My fault! i was looking at the pymata code base not the pymata3 code base!

closed 2017-03-21 by ajd394