MrYsLab/pymata-aio #55: Reconnecting, isConnected, or timeout exception?

Ahmed-90, 2017-09-10:
Hi, 

Is there any built-in way to detect board connection loss? 

i'm building a datalogger, i need to know in case the board died for whatever reason, like powerloss, cables failure or the board goes on fire :fire:

MrYsLab, 2017-09-10:
When a connection is detected as lost, pymata-aio exits. The reason being is that there is no easy way to reconnect and recover. You could use the keep_alive capability which will automatically reset an Uno or Leonardo (the Mega2560 does not have this capability) if there is a loss of communication, and if you set up the python script as an automatically restarting daemon, the system should be able to recover.

Ahmed-90, 2017-09-10:
@MrYsLab thanks for clarifying, the issue is that i'm using the `PyMata3` within a custom system where there is ton of other features, its not standalone daemon.

i was wondering if its possible to rise exception or a callback to notify the rest of the system that the board is down, so i can do a data cleanup, log the issue and send notification and try to re initialize the pymata

MrYsLab, 2017-09-11:
I put together a separate branch called **serial_exception** that has modified versions of pymata_core.py and pymata3.py. You are welcome to try it out to see if it meets your needs. I am not likely to merge these changes into the main branch since it breaks existing behavior for current users, but if it solves your problem, I would be willing to keep this branch updated with any future updates to the main branch.

Ahmed-90, 2017-09-11:
Thanks.. will give a try as soon as possible

MrYsLab, 2017-09-12:
I am closing this issue. Please feel free to leave additional comments or questions here.

closed 2017-09-12 by MrYsLab