MrYsLab/pymata-aio #57: latch example not working?

Ahmed-90, 2017-09-17:
i was trying to use `set_analog_latch`, but its didn't work.. i took a copy/paste of the wiki example still no luck.

can anyone confirm the analog latch is working correctly ?

MrYsLab, 2017-09-17:
I just ran the wiki example and it is working for me. What output do you get?
By the way, I am updating to version 2.18 in the next few minutes. Fixes some typos but no code changes.

MrYsLab, 2017-09-17:
Here is my output:
```
/usr/bin/python3.5 /home/afy/PycharmProjects/pymata-aio/examples/latching_test.py

pymata_aio Version 2.18	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

{}
['A2', 880, 1505680630.1086154]
Sun Sep 17 16:37:10 2017
{2: 880}
{2: 880}
```
The first entry is blank because there was no callback and "value" is not initialized. It is not a bug.

MrYsLab, 2017-09-17:
If you don't want to see the blank entry, change:
```
while True:
    print(value)
    board.sleep(5)
```

to:
```
while True:
    board.sleep(1)
    print(value)
    board.sleep(5)
```

MrYsLab, 2017-09-17:
You might also look at [test_all.py](https://github.com/MrYsLab/pymata-aio/blob/master/test/test_all.py) for other examples. All tests pass. Note that the pin numbers are hardcoded for the hardware I am using.

I run these tests in pycharm using the pytest test runner.

Ahmed-90, 2017-09-17:
thanks,  its not about the blank entry, 
i just tried again few sec ago and for some reason it worked
it used to give no output at all only `set_pin_mode` callback

anyway.. output:
```
{}
{2: 0}
['A2', 241, 1505680971.710847]
Sun Sep 17 23:42:51 2017
{2: 0}
{2: 530}
{2: 838}
{2: 0}
{2: 51}
{2: 460}
```

as you can see.. it only triggers once, when the signal drops below 99 and back up its never triggers again 
is that default behavior ?

Ahmed-90, 2017-09-17:
oh .. am blind didn't read the last line in the  the wiki page

MrYsLab, 2017-09-17:
That is correct behavior. The latch is a "one-shot" and must be rearmed after it fires. The [wiki page](https://github.com/MrYsLab/pymata-aio/wiki/Using-the-Latch-Data-Feature) states this at the bottom of the page.

MrYsLab, 2017-09-17:
No problem. I am going to close this issue.

closed 2017-09-17 by MrYsLab