MrYsLab/pymata-aio #59: pymata fails with ping

tim-nixon, 2017-10-30:
Servo works great, but ping never has the callback called
Console output (PC to genuine Uno)
c:\python363>python pingtest.py

pymata_aio Version 2.19 Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:com10

Initializing Arduino - Please wait...
Arduino Firmware ID: 2.5 StandardFirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

(here I hit ctl-c)
Traceback (most recent call last):
  File "pingtest.py", line 39, in <module>
    board.sleep(.1)
  File "c:\python363\lib\site-packages\pymata_aio\pymata3.py", line 542, in sleep
    self.loop.run_until_complete(task)
  File "c:\python363\lib\asyncio\base_events.py", line 454, in run_until_complete
    self.run_forever()
  File "c:\python363\lib\asyncio\base_events.py", line 421, in run_forever
    self._run_once()
  File "c:\python363\lib\asyncio\base_events.py", line 1390, in _run_once
    event_list = self._selector.select(timeout)
  File "c:\python363\lib\selectors.py", line 323, in select
    r, w, _ = self._select(self._readers, self._writers, [], timeout)
  File "c:\python363\lib\selectors.py", line 314, in _select
    r, w, x = select.select(r, w, w, timeout)
KeyboardInterrupt

code (directly from your site)
# import the API class
from pymata_aio.pymata3 import PyMata3

# ping callback function
def cb_ping(data):
    print(str(data[1]) + ' centimeters')

# create a PyMata instance
board = PyMata3(2)
# configure 4 pins for 4 SONAR modules
board.sonar_config(12, 12, cb_ping)

while True:
    board.sleep(.1)

board.shutdown()

MrYsLab, 2017-10-30:
It appears you are not using the Arduino sketch provided with this distribution. https://github.com/MrYsLab/pymata-aio/tree/master/FirmataPlus

Installation instructions can be found here:
https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino

Here is the output that I see when running the correct version of Firmata:

```
pymata_aio Version 2.19	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins
```
Pymata is fully compatible with StandardFirmata and apparently not with StandardFirmataPlus. I really wish that the StandardFirmata author did not name his library using a similar name that I had chosen 2 years prior to his coming out with StandardFirmataPlus. Perhaps he wasn't aware of my sketch.

In any case please the library indicated above. 

I am going to close this issue. If you have any other issues related to this matter please free to add another comment.

closed 2017-10-30 by MrYsLab

tim-nixon, 2017-10-31:
Thanks, that fixed it.. I had in fact picked the wrong one

Question: the ping seem limited to close distances, but the ping sketch to maybe 15 feet. Is that a limitation of firmataplus, ptfirmata or a combination?

MrYsLab, 2017-10-31:
Glad that it is working. I don't have a definitive answer as for the limitation, but it may be a limitation in the NewPing library that my sketch uses. Also, it depends if there are reflections occurring off of other objects including walls and the size of the target object.

tim-nixon, 2017-11-01:
It appears that it is firmataplus. or the interface from pymata. If I change the line in firmataplus from
sonars[numActiveSonars] = new NewPing(sonarTriggerPin, sonarEchoPin, max_distance) ;
and substitute 1000 for max_distance, I can read out to 150 inches..  not sure why that is true

MrYsLab, 2017-11-01:
@tim-nixon Throttling the distance to 200 cm was brought over from PyMata, and I am not really sure if that throttling is necessary for pymata-aio. The maximum range for[ HC-SR04 is 400 cm ](https://elecfreaks.com/estore/download/EF03085-HC-SR04_Ultrasonic_Module_User_Guide.pdf)which is about 150 inches.  If you would like to try to extend that range to the maximum of 400 cm you can modify lines 1156 and 1157 of pymata_core.py from:
```
if max_distance > 200:
            max_distance = 200
```
to

```
if max_distance > 400:
            max_distance = 400
```

If it works you can either enter a pull request or let me know and I can make the changes. I prefer to keep the default value to 200 for legacy purposes, but am willing to remove the 200 cm throttle.

Let me know.