MrYsLab/pymata-aio #62: different behaviour when running under VS code and from bash

jara1, 2018-03-22:
Hi,

I'm new to pymata and trying to get working pymata-aio under Linux Mint (running in Virtualbox). Python 3.5.1. Using example (https://gist.github.com/MrYsLab/8b735813e413bf62b455).

When I run it in Visual studio code debugger it behaves better than when running from bash. See attached logs and scope captures (green - TX from PC, yellow RX to PC). It seems to me, that when running from bash the script does not wait for response and is transmitting commands too fast.

[log.txt](https://github.com/MrYsLab/pymata-aio/files/1837457/log.txt)

VS code:
[image]

Bash:
[image]

Thanks.
JC

MrYsLab, 2018-03-22:
Try changing line 13 from:
```
board = PyMata3()
```

to
```
board = PyMata3(arduino_wait=5)
```
Or some other wait time value. This time is how long pymata-aio waits for the Arduino to reset itself.
The default is 2 seconds.

Please let me know if this solves your problem.

jara1, 2018-03-22:
No, no change in behaviour. BTW arduino responds in less than **2ms** to the second command (probably get version command). So that's not the problem.

MrYsLab, 2018-03-22:
I am not clear what the problem is that you are experiencing.
Is it that it is timing out as shown in your text file? Btw I run ubuntu and can't recreate the timeout.

jara1, 2018-03-22:
There are probably two independent problems. The first (and lower priority) is the error in VS code (`Task was destroyed but it is pending!`).
The second is the timeout. Lets focuse on the timeout for now. Based on the error it's not timeout to the reset or version command, it's some subsequent command to get analog map. To me it seems that command are quite close to each other (without relation to the responses, like they are asynchronous to the responses).
[image]

MrYsLab, 2018-03-22:
Since I am not able to reproduce the problem here, can you tell me which version of mint you are running? Is it 18.3? Also, are you running 32 bit or 64 bit and is it mate, xfce or kde?

I will install on VBox and will see if that allows me to reproduce the problem.

MrYsLab, 2018-03-22:
One other question - what Arduino board are you using? Is it an official arduino?

MrYsLab, 2018-03-22:
I just loaded Mint and cannot reproduce your issue. My host system is Ubuntu 17.10 and the guest is the same version of Mint that you are using. I am using an Arduino Leonardo.

One thing you can try is to run the same script without the Pymata3 wrapper:

```
async def pwm(my_board):
    """
    Set digital pin 6 as a PWM output and set its output value to 128
    @param my_board: A PymataCore instance
    @return: No Return Value
    """
    # set the pin mode
    await my_board.set_pin_mode(13, Constants.PWM)

    await board.analog_write(13, 128)
    # set the pin to 128
    await my_board.analog_write(13, 128)

    await asyncio.sleep(3)

    # shutdown
    await my_board.shutdown()

board = PymataCore(5)

# get the loop, and execute pin6_pwm_128
loop = asyncio.get_event_loop()
loop.run_until_complete(board.start_aio())
loop.run_until_complete(pwm(board))

```
If that does not work for you, I do not know how to proceed since I am unable to reproduce your problem nor has it been reported before.

Note that I am using the pin13 LED and not pin 6 for this program, but that should not matter.

jara1, 2018-03-23:
Thank you for inevstigation.
I tested it without wrapper and the same issue.
I tested fresh instalation of xubuntu (in virtualbox), the same issue.
I tested it on native win7 - works without issue.
I will investigate it more.

MrYsLab, 2018-03-23:
Some additional information. I updated my version of VB on my Windows 10 laptop to the latest. I am running Ubuntu 17.10 and Python 3.6 and all runs correctly. So I am not sure if your problem is something unique to your environment or if Windows 7 is the problem. 

I am closing this issue, but please feel free to add a comment and I will see it even though the issue is closed.

closed 2018-03-23 by MrYsLab