MrYsLab/pymata-aio #63: How can I make a unit test for board correction?

JosephMart, 2018-03-29:
How do you suggest that I create a unit test for validation a proper connection to an Arduino board?

MrYsLab, 2018-03-29:
There are 2 possibilities to test for:
1. The board is not plugged.
2. The board is plugged in, but the incorrect sketch is loaded on Arduino.

Is this what you are looking for? 

```
import asyncio
from pymata_aio.pymata3 import PyMata3

def test_connection():
    try:
        board = PyMata3()
    # Board not plugged in - correct sketch or not
    except TypeError:
        assert False
    # incorrect sketch loaded
    except asyncio.futures.CancelledError:
        assert False
```

JosephMart, 2018-03-31:
Yep this is what I wanted. Thanks!

closed 2018-03-31 by JosephMart