MrYsLab/pymata-aio #64: burn hex and reconnect arduino

ouki-wang, 2018-04-03:
My goal is to connect this board. After an exception occurs, I can use python to burn firmata.hex and then connect again.
Reconnect again with the following error：

using com COM31

pymata_aio Version 2.19	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Initializing Arduino - Please wait... 
Arduino Firmware ID: None
*** Analog map retrieval timed out. ***

Do you have Arduino connectivity and do you have a Firmata sketch uploaded to the board?

Shutting down ...
An exception occurred on the asyncio event loop while receiving data.  Invalid message.

avrdude.exe: Version 6.3, compiled on Mar 27 2018 at 09:51:57
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Program Files (x86)\Scratch\Arduino/hardware/tools/avr/etc/avrdude.conf"

             Using Port                    : COM31
             Using Programmer              : arduino
             Overriding Baud Rate          : 115200
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 3
             Firmware Version: 4.4
             Vtarget         : 0.3 V
             Varef           : 0.3 V
             Oscillator      : 28.800 kHz
             SCK period      : 3.3 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading input file "C:\Program Files (x86)\Scratch\Arduino/fw/uno/StandardFirmata.hex"
avrdude.exe: writing flash (14464 bytes):

Writing | ################################################## | 100% 2.20s

avrdude.exe: 14464 bytes of flash written
avrdude.exe: verifying flash memory against C:\Program Files (x86)\Scratch\Arduino/fw/uno/StandardFirmata.hex:
avrdude.exe: load data flash data from input file C:\Program Files (x86)\Scratch\Arduino/fw/uno/StandardFirmata.hex:
avrdude.exe: input file C:\Program Files (x86)\Scratch\Arduino/fw/uno/StandardFirmata.hex contains 14464 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 1.70s

avrdude.exe: verifying ...
avrdude.exe: 14464 bytes of flash verified

avrdude.exe done.  Thank you.

C:\Program Files (x86)\Scratch\Arduino/hardware\tools/avr/bin/avrdude.exe
-CC:\Program Files (x86)\Scratch\Arduino/hardware/tools/avr/etc/avrdude.conf
-v -patmega328p -carduino -PCOM17 -b115200
-Uflash:w:C:\Program Files (x86)\Scratch\Arduino/fw/uno/StandardFirmata.hex:i

pymata_aio Version 2.19	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Initializing Arduino - Please wait... Event loop stopped before Future completed.
C:\Program Files (x86)\Scratch\Arduino/hardware\tools/avr/bin/avrdude.exe
-CC:\Program Files (x86)\Scratch\Arduino/hardware/tools/avr/etc/avrdude.conf
-v -patmega328p -carduino -PCOM17 -b115200
-Uflash:w:C:\Program Files (x86)\Scratch\Arduino/fw/uno/StandardFirmata.hex:i

avrdude.exe: Version 6.3, compiled on Mar 27 2018 at 09:51:57
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Program Files (x86)\Scratch\Arduino/hardware/tools/avr/etc/avrdude.conf"

             Using Port                    : COM31
             Using Programmer              : arduino
             Overriding Baud Rate          : 115200
avrdude.exe: ser_open(): can't open device "\\.\COM31": Access is denied.

avrdude.exe done.  Thank you.

C:\Program Files (x86)\Scratch\python36\lib\site-packages\pymata_aio-2.19-py3.6.egg\pymata_aio\pymata_core.py:1112: RuntimeWarning: coroutine 'PymataSerial.close' was never awaited
Task exception was never retrieved
future: <Task cancelling coro=<PymataCore._command_dispatcher() done, defined at C:\Program Files (x86)\Scratch\python36\lib\site-packages\pymata_aio-2.19-py3.6.egg\pymata_aio\pymata_core.py:1284> exception=RuntimeError('This event loop is already running',)>
Traceback (most recent call last):
  File "C:\Program Files (x86)\Scratch\python36\lib\site-packages\pymata_aio-2.19-py3.6.egg\pymata_aio\pymata_core.py", line 1297, in _command_dispatcher
    next_command_byte = await self.read()
  File "C:\Program Files (x86)\Scratch\python36\lib\site-packages\pymata_aio-2.19-py3.6.egg\pymata_aio\pymata_serial.py", line 155, in read
    await asyncio.sleep(self.sleep_tune)
  File "C:\Program Files (x86)\Scratch\python36\lib\asyncio\tasks.py", line 482, in sleep
    return (yield from future)
concurrent.futures._base.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Scratch\python36\lib\site-packages\pymata_aio-2.19-py3.6.egg\pymata_aio\pymata_core.py", line 1350, in _command_dispatcher
    loop.run_until_complete(asyncio.sleep(.1))
  File "C:\Program Files (x86)\Scratch\python36\lib\asyncio\base_events.py", line 454, in run_until_complete
    self.run_forever()
  File "C:\Program Files (x86)\Scratch\python36\lib\asyncio\base_events.py", line 408, in run_forever
    raise RuntimeError('This event loop is already running')
RuntimeError: This event loop is already running

MrYsLab, 2018-04-03:
I don't understand what your program does. I noticed that the baudrate being reported is 115200. Firmata expects 57600. If you are trying to integrate avrdude with pymata-aio, you may be mixing asyncio with non-asyncio code which can be difficult to do.

i

MrYsLab, 2018-04-03:
If you are not combining asyncio and non-asyncio, then try setting the arduino_wait parameter to some high value like 30. If that works, adjust it down to a smaller value.

MrYsLab, 2018-04-04:
I am closing this issue since I do not believe your problem is being caused by pymata-aio. If you disagree, please provide some detailed information about your application (like the source code, instructions on how to use it, etc.), and I will be happy to reopen and investigate your issue.

closed 2018-04-04 by MrYsLab