MrYsLab/pymata-aio #67: RuntimeError: This event loop is already running

troymessina, 2018-06-07:
I have installed pymata-aio on four different Windows machines. Two are Windows 7, and two are Windows 10. I get the error "RuntimeError: This event loop is already running" on one of the Windows 7 and one of the Windows 10 machines, but everything is fine on the other two machines.   I am running in a Jupyter notebook managed by Anaconda (Python 3.5 and 3.6) and have tried lots of different examples from your wiki and some that I wrote. I have tried hard-coding the COM port, changing the arduino_wait, restarting the machines, restarting the Python kernal, unplugging/replugging the Arduino into the USB. Nothing will resolve the issue.  Is there something I am missing? I've included an example below.

`"""
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc
"""

from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants

BOARD_LED = 13
board = PyMata3(arduino_wait=5)

def setup():
    board.set_pin_mode(BOARD_LED, Constants.OUTPUT)

def loop():
    print("LED On")
    board.digital_write(BOARD_LED, 1)
    board.sleep(1.0)
    print("LED Off")
    board.digital_write(BOARD_LED, 0)
    board.sleep(1.0)

if __name__ == "__main__":
    setup()
    while True:
        loop()`

`pymata_aio Version 2.20	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:com6

Initializing Arduino - Please wait... 
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-1-06a41e5573ed> in <module>()
     12 
     13 BOARD_LED = 13
---> 14 board = PyMata3(arduino_wait=5)
     15 
     16 

~\AppData\Local\Continuum\Anaconda3\lib\site-packages\pymata_aio\pymata3.py in __init__(self, arduino_wait, sleep_tune, log_output, com_port, ip_address, ip_port, ip_handshake)
     60         self.core = PymataCore(arduino_wait, self.sleep_tune, log_output,
     61                                com_port, ip_address, ip_port, ip_handshake)
---> 62         self.core.start()
     63         self.sleep(1)
     64 

~\AppData\Local\Continuum\Anaconda3\lib\site-packages\pymata_aio\pymata_core.py in start(self)
    290         asyncio.ensure_future(self.get_firmware_version())
    291 
--> 292         firmware_version = self.loop.run_until_complete(self.get_firmware_version())
    293         if self.log_output:
    294             log_string = "\nArduino Firmware ID: " + firmware_version

~\AppData\Local\Continuum\Anaconda3\lib\asyncio\base_events.py in run_until_complete(self, future)
    453         future.add_done_callback(_run_until_complete_cb)
    454         try:
--> 455             self.run_forever()
    456         except:
    457             if new_task and future.done() and not future.cancelled():

~\AppData\Local\Continuum\Anaconda3\lib\asyncio\base_events.py in run_forever(self)
    407         self._check_closed()
    408         if self.is_running():
--> 409             raise RuntimeError('This event loop is already running')
    410         if events._get_running_loop() is not None:
    411             raise RuntimeError(

RuntimeError: This event loop is already running
`

MrYsLab, 2018-06-07:
I just installed anaconda on a fresh installation of Linux. If I run blink.py from a terminal window, it runs without error. If I bring up a web-based jupyter notebook and run blink.py, I see the same error you are seeing.

If you run outside of the jupyter notebook does it run successfully for you?

It appears that this is a bug within jupyter. I wanted to try and see if I can debug, but spyder crashes on my system.

MrYsLab, 2018-06-07:
Yet more info. I am pretty sure now that this is a jupyter notebook issue. Take a look at this jupyter issue:
https://github.com/jupyter/notebook/issues/3397.

I am going to close this issue, but if you still think this is a pymata-aio issue, you can add a comment, and I will be happy to reopen the issue.

closed 2018-06-07 by MrYsLab

MrYsLab, 2018-06-07:
Here's the fix. You need to regress the version of a Tornado in jupyter notebook as per the jupyter issue above

```
pip install tornado==4.5.3
```

troymessina, 2018-06-07:
That does the trick. Thank you!