MrYsLab/pymata-aio #70: INPUT_PULLUP mode clashes with SONAR

toxuin, 2018-07-04:
I may be missing something here, but looking at Arduino's Firmata library, flag 0x0b is pin mode `INPUT_PULLUP`. In this library, however, it is reserved for `SONAR`.

In my code I was able configure pin correctly just by using `0x0b`, but such discrepancy is confusing.

Source: [Arduino Firmata library constants](https://github.com/firmata/arduino/blob/74980b9fe49c7ed9444acc6d4aef04ea8b16ad29/FirmataConstants.h#L79)

MrYsLab, 2018-07-04:
@toxuin Thanks for letting me know about this. I am surprised that no one (including me) found this before. I'm working on a fix, but it requires a lot of testing (FirmataPlus and all of its flavors, and compatibility with PyMata). As soon as it is ready, I will add a comment here.

MrYsLab, 2018-07-04:
Version 2.22 should fix the problem. Please let me know if you have any further issues or questions.

closed 2018-07-04 by MrYsLab

toxuin, 2018-07-04:
Hey, thanks for such quick response & release!