MrYsLab/pymata-aio #79: pymata win10

petrkorda, 2018-09-11:
PS C:\Users\Petr> s2aio -c no_client

pymata_aio Version 2.22 Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:com3

Initializing Arduino - Please wait... list index out of range
Shutting down ...

I ve installed this on win7 and raspbian, it's OK. In windows 10 :( Gives this error

MrYsLab, 2018-09-11:
Since invoking s2aio caused the error, let's try to isolate if you are in general having a problem with pymata-aio.  First, do you see this error if you start from a power-up situation - that is, the Arduino has been repowered by removing and replacing the USB cable? If you restart a program (including s2aio) after configuring the Arduino, let's say for analog input, this type of error can appear since there is no way to fully reset the Arduino and data continues to restream into pymata-aio.  

If it is happening even from a cold start, could you please try running the following program. It will blink the Arduino LED attached to pin 13 and print out some state information to the console. Please let me know if that works.
```
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants

BOARD_LED = 13
board = PyMata3()

def setup():
    board.set_pin_mode(BOARD_LED, Constants.OUTPUT)

def loop():
    print("LED On")
    board.digital_write(BOARD_LED, 1)
    board.sleep(1.0)
    print("LED Off")
    board.digital_write(BOARD_LED, 0)
    board.sleep(1.0)

if __name__ == "__main__":
    setup()
    while True:
        loop()

```
Here what the output should look like:
[image]

This was run on Windows 10.

MrYsLab, 2018-09-14:
Were you able to try the blink code? Are you still having issues?

MrYsLab, 2018-09-16:
I am going to close this issue since I have not heard back from you. If you are still having issues and wish to reopen this item, please leave a comment here and we can pursue any problems you are having.

closed 2018-09-16 by MrYsLab

petrkorda, 2018-09-26:
Sorry for my late response. i´m using s2aio with scracth 2.
I´ve found the problem. The problem is i have more than a COM port im computer, i guess, and win10 is more stupid, and dont get the right one. To work i have to specify the correct one (example: s2aio -p COM5).
In win 7 it´s and raspbian it detect´s well (it is auto).
s2aio server is stopping after  changing files on scracth 2, i'm solving it using more time: s2aio -p COM5 -s 50
Thx for you work.

______________________________________________________________________________________________
casaldanarceja.wordpress.com<http://casaldanarceja.wordpress.com>
Aprenda o verdadeiro office em: https://www.youtube.com/c/ComunidadeLibreOfficePortugal
<http://casaldanarceja.wordpress.com>
Paulo Galvão

[image]<https://user-images.githubusercontent.com/5189838/45370333-05c0d900-b5b6-11e8-8117-7d63fa67047b.png>

MrYsLab, 2018-09-26:
Thank you for letting me know.

petrkorda, 2018-10-11:
Good Afternoon, it's me again because the sh... of window$ 10.
I was trying to use PyMata3 but it didt´n work until i put board = PyMata3(arduino_wait=5), as a suggestion of another user.

Can you put a comment in code #put this in windows PyMata3(arduino_wait=5)? I lost 4 hours to find the solution. The strange thing is that Pymata works fine.
https://github.com/MrYsLab/pymata-aio/blob/master/examples/blink.py
[https://avatars3.githubusercontent.com/u/5189838?s=400&v=4]<https://github.com/MrYsLab/pymata-aio/blob/master/examples/blink.py>

pymata-aio/blink.py at master · MrYsLab/pymata-aio · GitHub<https://github.com/MrYsLab/pymata-aio/blob/master/examples/blink.py>
This is the second generation PyMata client. Contribute to MrYsLab/pymata-aio development by creating an account on GitHub.
github.com

Thanks.

______________________________________________________________________________________________
casaldanarceja.wordpress.com<http://casaldanarceja.wordpress.com>
Aprenda o verdadeiro office em: https://www.youtube.com/c/ComunidadeLibreOfficePortugal
<http://casaldanarceja.wordpress.com>
Paulo Galvão

MrYsLab, 2018-10-11:
I just updated the code with some additional comments. I am leaving the default arduino_wait to 2 seconds since it works for most Windows (as well as other OS's) computer. There is a comment explaining this.

I think the problem you are having is unique to your computer/arduino setup. It could be the result of the processing speed of your computer, or the type of Arduino you are using(official versus clone), the USB serial driver, or a combination of any of these.

I hope this helps.

petrkorda, 2018-10-24:
firmataplus give this errors but compile ("portable" arduino in raspberrypi)

works ok: blink.py, servo_sweep.py, core_sample.py, keyboard_notes... testing.

/home/pi/Downloads/arduino-1.8.7/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:48:29: warning: extra tokens at end of #include directive
 #include <ooPinChangeInt.h> https://code.google.com/p/arduino-pinchangeint/downloads/detail?name=pinchangeint-v2.19beta.zip
                             ^
/home/pi/Downloads/arduino-1.8.7/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:49:25: warning: extra tokens at end of #include directive
 #include <AdaEncoder.h> https://code.google.com/p/adaencoder/downloads/detail?name=adaencoder-v0.7beta.zip
                         ^
/home/pi/Downloads/arduino-1.8.7/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino: In function 'void sysexCallback(byte, byte, byte*)':
/home/pi/Downloads/arduino-1.8.7/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino:849:36: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
           printData("argc = ", argc) ;
                                    ^
/home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/AdaEncoder.cpp:41:0: warning: "cli" redefined
 # define cli()  __asm__ __volatile__ ("cli" ::)
 ^
In file included from /home/pi/Downloads/arduino-1.8.7/hardware/arduino/avr/cores/arduino/Arduino.h:30:0,
                 from /home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/../ooPinChangeInt/ooPinChangeInt.h:113,
                 from /home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/AdaEncoder.h:57,
                 from /home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/AdaEncoder.cpp:29:
/home/pi/Downloads/arduino-1.8.7/hardware/tools/avr/avr/include/avr/interrupt.h:99:0: note: this is the location of the previous definition
 # define cli()  __asm__ __volatile__ ("cli" ::: "memory")
 ^
/home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/AdaEncoder.cpp:42:0: warning: "sei" redefined
 # define sei()  __asm__ __volatile__ ("sei" ::)
 ^
In file included from /home/pi/Downloads/arduino-1.8.7/hardware/arduino/avr/cores/arduino/Arduino.h:30:0,
                 from /home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/../ooPinChangeInt/ooPinChangeInt.h:113,
                 from /home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/AdaEncoder.h:57,
                 from /home/pi/Downloads/arduino-1.8.7/libraries/AdaEncoder/AdaEncoder.cpp:29:
/home/pi/Downloads/arduino-1.8.7/hardware/tools/avr/avr/include/avr/interrupt.h:81:0: note: this is the location of the previous definition
 # define sei()  __asm__ __volatile__ ("sei" ::: "memory")
 ^

MrYsLab, 2018-10-24:
Those warnings are safe to ignore, but if you would like not to see them I uploaded a new l[ibraries.zip ](https://github.com/MrYsLab/pymata-aio/blob/master/FirmataPlus/libraries.zip) that should get rid of all the warnings when the Arduino IDE is set to default compiler warnings. The warning level can be set from File/Preferences dialog in the IDE.

Please reinstall the Firmataplus files using the new library and let me know if that solves your problem.
Thanks