MrYsLab/pymata-aio #80: OneWire

voidloop, 2018-10-11:
It would be great to have OneWire support for ConfigurableFirmata

https://github.com/firmata/ConfigurableFirmata/blob/master/src/OneWireFirmata.h

MrYsLab, 2018-10-12:
I am going to mark this as an enhancement request. I will not be supporting ConfigurableFirmata directly, since in the past it was very buggy and difficult to work with. What I would be willing to try is to merge the SPI support into FirmataPlus.
The only thing preventing me from doing this is that I have no SPI devices and have no way of testing.

When I finish up my latest projects, I will explore if there are any reasonably priced SPI devices available and then try to integrate. I have no time frame as to when this might be.

MrYsLab, 2018-10-12:
I just did a search on Sparkfun and Adafruit for SPI devices and it seems the ones they carried are discontinued. Do you have a specific device in mind?

voidloop, 2018-10-12:
Thank you. 
No any device for SPI, but for 1-Wire I have in mind the DS18B20, a temperature sensor.

MrYsLab, 2018-10-12:
You are correct - sorry for my confusion. Since ConfigurableFirmata implements this as a bit-banger application, I am not sure that it can be integrated into FirmataPlus which is based on StandardFirmata.

One of the rationales (I am not sure if this still is the case) for ConfigureableFirmat was to support devices that would interfere with the timing of the main loop of StandardFirmata. This means, that certain configurable components were to be made available in a standalone configuration, or where the user was not concerned about the main loop timing being altered in an unpredictable way by adding the configurable feature.

When I explore this, if this is the case with onewire, then most likely I will not be able to integrate it into FirmataPlus. Pymata assumes all features are available and can coexist without interfering with each other.

When I get to explore this in greater detail, I will let you know.

MrYsLab, 2018-10-30:
@voidloop I just ordered a DS18B20. I will keep you informed.

voidloop, 2018-11-04:
Hi, I've tried my DS18B20 with the library you've linked and it works, I've use it with a Wemos D1 mini pro v.1.0.0. So, the library meets my needs.
Have you already uploaded the pymata-aio example?
Thank you again.

MrYsLab, 2018-11-04:
I have researched supporting onewire and what I found is not encouraging.
After investigating the ConfigureableFirmata solution, I don't believe that it is consistent enough.
Here is why:
1. They provide for searching for alarms, but no means to set alarm thresholds in the device.
2. They provide a generic onewire interface, but the view, IMHO is too low level. For example, looking at one of the two [ Firmata client I could find for onewire](https://github.com/openframeworks/openFrameworks/blob/patch-release/libs/openFrameworks/communication/ofArduino.cpp),  they had to add code to make things more user-friendly. 
3. The other client, [johnny-five](https://github.com/rwaldron/johnny-five) simply reads the temperature and does not handle alarms.

There are 3 possible libraries that support the Arduino, and none seem to be a total solution. 

The [teensy library] (https://www.pjrc.com/teensy/td_libs_OneWire.html)  does not directly support alarm search or setting alarm thresholds.
The [Dallas Temperature Library](https://github.com/milesburton/Arduino-Temperature-Control-Library) is specific to the temperature devices, but not other onewire devices.
And as I mentioned above, the ConfigureFirmata library is too low level and incomplete.

So if you are looking for support only for the DS18B20, I would consider possibly providing it. If you are looking for support Onewire devices in general, then I will have to say that I will not be providing it, due to development time, cost of purchasing different devices and the time to test these devices.

I am closing this issue for now, Please let me know what type of support you are looking for, I can reopen it if necessary.

MrYsLab, 2018-11-04:
Looking at your comment above mine (I did not see your comment until now), here is what I would be willing to provide:
Support for a single DS18B20 per pin and not generic onewire support.
Would this meet your needs?

voidloop, 2018-11-06:
> Looking at your comment above mine (I did not see your comment until now), here is what I would be willing to provide:
> Support for a single DS18B20 per pin and not generic onewire support.
> Would this meet your needs?

Yes. I need read temperature from two sensors, if I can use sensors each to one pin it's OK.
I don't want to bore you especially I don't want that you waste your time solving a problem only for me, so if using a device with onewire protocol is the hard way on Firmata, it isn't a problem for me change my project. Let me know and thank you!

MrYsLab, 2018-11-06:
Thank you for letting me know. I have things partially working and hope to have things fully working soon.
The code should be able to work with multiple devices on a single pin or on separate pins. 
Thanks.

MrYsLab, 2018-11-07:
Yet another update. I looked at supporting alarms but decided I will not implement this feature. The alarms are not latched within the DS18B20 and to add an alarm handler for each DS18B20 within the Firmata sketch just adds a lot of complication. 

The user can simply check for exceeding either a low or high threshold in their call back routine and that will simplify things quite a bit.

I will be documenting the feature on the Wiki page, and need to clean up my code and do some more testing. I need to add a working example for both pymata_core and Pymata3. I have a working example where I am continuously reading the temperature while simultaneously monitoring a potentiometer that sets a PWM level on an LED.  Seems to be working for Pymata3, and I just need to create a pure asyncio version for pymata_core.

Hopefully, I can release within the next several days.

MrYsLab, 2018-11-08:
Well, I have some bad news. I thought I had a working test program that reads the DS18B20 while simultaneously reading an analog input and using that value to set the value on PWM LED. The temperature read works well, but it is blocking firmata from reading the analog input. When I tested orginially, I had a bug that did not use the DS18B20 arduino library correctly, and when I fixed that issue, the library blocks FirmataPlus. I still get accurate reads, but the analog read is being blocked (I never see a value change when I move the potentiometer).

I ran a test without Firmata and just modified examples provided by the onewire library and the Dallas library to do the same as above, and it also blocked things. So it seems to be inherent with onewire.

I have created a branch called onewire on Github that contains FirmataPlusDS, 2 examples (ds_tests.py and pymata3_ds18b20.py) and some updated files.

If you wish you may try that branch to see if it meets your needs. If you need any help setting it up, please let me know, and I will do so.

Because of the blocking issues, I will not release the code as part of an "official" release.

I appreciate your request, and I learned a lot about onewire devices, so it was not a wasted effort.

I will close this issue. If you have any comments, you can post them here even though the issue is closed and I will see them.

Thanks again.

MrYsLab, 2018-11-08:
I am reopening the issue since I want to try another approach. The reason for the blocking of Firmata was that I was waiting for the temperature conversion to complete within the sketch. This was the blocking action.

What I want to experiment with is to issue a request to the device for conversion, but not wait for the completion, but to continue on processing and sometime in the future come back to retrieve the result of the conversion.

Doing this will impose some limitations on the user. First, I am going to limit the total number of sensors handled to 10. The sensors may be distributed across multiple pins or assigned to a single pin or any combination the user wishes. All of the devices will start a conversion at the "same" time within the FirmataPlus loop and all of the results will also be collected together after the loop executes for some period of time. What this means is that the user will not control when the conversion period starts or the results are returned. This will be controlled by the FirmataPlus sketch.

I am hoping to provide a way for the user to have the ability to stop and restart any given sensor.

Once again, I will provide updates as they are available. I apologize for all the confusion, but I am trying to get something that will meet your needs and that I can add to the "official" release.

MrYsLab, 2018-11-09:
I was unable to make this latest plan work, so I am permanently closing this issue. I apologize for all the confusion and appreciate your patience.

closed 2018-11-09 by MrYsLab

voidloop, 2018-11-13:
Thank you anyway, you were very gentle.