MrYsLab/pymata-aio #81: calls to exit() rather than sys.exit() in pymata_core.py

ericwertz, 2018-10-14:
Note: I previously, mistakenly submitted this bug against "pymata" rather than here in "pymata-aio". 
 I think/hope that I've closed that bug report out... but perhaps not.  Apologies.
----------
During the course of some trying to track down some almost-exclusively Windows-specific runtime problems, I've hit one or more paths where exit() is being called, rather than what should be sys.exit() in pymata_core. A quick search finds that there are two such calls currently in pymata_core.py at lines 990 and 1869.

Since we're talking about calls to sys.exit(), there are numerous places where sys.exit(0) is called in non-happy paths that seem to be runtime errors, but for which no diagnostic messages are issued. Any of these that are hit would result in silent deaths, making them hard to track down if they're hit. At least a print() would help, and a full-on runtime error w/stack trace might even be more helpful in some of these contexts.

MrYsLab, 2018-10-14:
Any place where exit is called is in a location where unrecoverable occurred.

Let me comment on all of the places where exit is called:
line 287 - a print statement is generated.

lines 330 and 332 - there is no connectivity to the Arduino and the code is trying to close down asynio just above that. A print statement is generated in line 316.

line 395 generates a print statement.

lines 426, 432 and 434 a print statement is generated just above.

lines 464, 470 and 472 a print statement is generated just above.

line 990 -  The only time that I am aware of that an exception is generated as a result of when shutdown() calls send_reset. If send_reset is called on its own and there is a problem in the call to _send_command, _send_command will issue a print statement. If shutdown is called, a print is generated.

line 1164 - there is a print statement above.

line 1427 - there is a print statement above.

line 1869 - there is a print statement above.

So, are you seeing unidentifed exits and if so, please let me know the scenario so I can try to reproduce it here.

Thanks

ericwertz, 2018-10-14:
Alan,
    I hate to have seen you spend so much time on this response.  The ones that jumped out were around line 330.  One possible exception presumes that the RuntimeError exception can be due only to one particular error, and the TypeError was the one I was actually most worried about.
    I will absolutely concede that you know what's more likely to be correct and more helpful to diagnose than, so if you feel that any third-person can ID the reason for a shutdown through these paths, then feel free to ignore what I've pointed out from just a desk review.
    What I *have* seen are Python runtime syntax errors that state that there's no "exit()" in the namespace when that call is hit.  This is the immediate concern, and my other observations were only through a desk-review while I was trying to track down the unqualified calls to exit().
    As you already know, I'm trying to track down the source of a very high number of failures on ~50 different students' (roughly 50% of) Windows laptops on which were trying to run PyMata3().
    Thanks *very* much for your time.

best regards,
-eric

ericwertz, 2018-10-14:
> So, are you seeing unidentifed exits and if so, please let me know the scenario so I can try to reproduce it here.

Sorry that I didn't address this Q of yours directly.  I detailed the problem symptoms that I'm seeing in a separate email to MrYsLab@gmail.com last night.  Hitting this particular naked exit() call is something that I've seen first-hand numerous times during the course of working with these students' machines.  I don't have a stack trace that I can provide (yet) but I'll try to get one.  Since I can't reproduce any of the crashes or bare exit()s on my machine, I can't readily provide one.

This particular error may be the result of going down a path where the COM is being held open by a un-terminated previous run that hasn't shut down completely (before doing a kernel reset in JupyterLab), and where I don't *believe* that I've seen explanatory print() output.  I will confirm that this really is a silent exit the next time I see it.  If this particular error is the result of the COM port already being busy, I can certainly reproduce this scenario on my own machines which otherwise run PyMata3 near-flawlessly.

best regards,
-eric

MrYsLab, 2018-10-14:
I should probably change the exits to sys.exit for consistency. I will open up an issue so that I do so the next time I make a change.

I just ran the 2 following scripts both on Linux and Windows and both behave the same. I also ran a code inspection on both scripts in PyCharm and it does a strict check for any problems. None reported.

```
import time

print('hello')
time.sleep(1)
print('goodbye')
exit(0)

```
```
import time
import sys

print('hello')
time.sleep(1)
print('goodbye')
sys.exit(0)

```

I am going to close the issue, but feel free to add any comments and if this should reopened I will do so.

closed 2018-10-14 by MrYsLab

ericwertz, 2018-10-14:
I know nothing about the particulars of a shutting-down Python runtime, but I do know that I've seen errors of calls to exit() that can't be found in the namespace.  Understandable given that's it's an obscure non-happy path and likely isn't a negative-test case that is hit in any automated regression testing that you may be doing.  I can't account for why PyCharm's sounds-like-static-analysis tool didn't find this.  I haven't started to use PyCharm yet, but it seems imminent.  I'm pretty new to Python, so my tooling situation is negligible at this instant in time.

best regards,
-e

MrYsLab, 2018-10-14:
@ericwertz I just opened issue #82 to track this.

Could you please remind about the failures on your student's laptops. Do they all fail in the same way?
If so, what is the failure again?

In issue #79 a user reported some problems. When he set the arduino_wait parameter to 5 seconds, things worked for him. I don't remember, did we try doing that for your students?

MrYsLab, 2018-10-14:
I can highly recommend PyCharm. I used to be a "debug by print guy", but having the IDE has made things some much easier and efficient. 

You can get a free copy of their professional edition for both you and your students:
https://www.jetbrains.com/student/

One advantage of the professional version is that you can run a code on a remote python interpreter and debug it on your pc. I do that all the time when working with the Raspberry Pi. Makes life much easier ;-).

MrYsLab, 2018-10-14:
One more plug for Pycharm - if you get the student license, you also get access to all of their other products. The user interface is pretty much consistent across their product line, so you only have to go through the pain once. I have used WebStorm, IntelliJ (Java), and RubyMine. All work well.

ericwertz, 2018-10-14:
Alan,

Fuh... well that explains it.  I just found in my Junk folder that Gmail bounced-back my original message to you yesterday.  That explains the lack of context.  I apologize for that.  Apparently I didn't get your email addressed properly across FingerNet.

I'm cutting-and-pasting the original note along with the single image attachment from that note.  Sorry again that you didn't get the original.  I hope this explains everything, but please let me know if it doesn't.

Grrrr, I'm having trouble attaching the aforementioned image to this email -- HotMail is *really* acting up right now.  I'll try to send it immediately separately as a parallel response to this note.

May I ask which timezone you're in?

best regards,
-eric

-------

Subject: 50% failure rate on Windows ("only") with PyMata3
From: Eric Wertz
Date: Sun 10/14/2018, 12:01 AM

Alan,
    The professor with whom I'm working at San Jose State University (burford.furman@sjsu.edu) earlier this year had a question for you about our use of PyMata3 within Anaconda/Jupyterlab which you were able to help with.  For the four of us using your platform on Windows since then to develop our courseware since then, we've had very few problems.
    Unfortunately, this week we started rolling the platform out to students and our failure rate has been substantial.  12/13 MacOS users were able to get up and running with no issues.  The other one has the more common problem of the two that's plaguing 50% of our  (~70) Windows users, who can't get our simple blink example to come up properly.  We're using PyMata3 and the pymata-aio comms implementation.
    The two error symptoms on Windows are both within the call to PyMata3().  In all cases, the COM port seems to be opened successfully.
    The majority's error is a Python runtime error that "firmware_version" is None when it's trying to be printed out during the initial handshake phase, so I'm guessing that there's either a non-happy path that isn't setting it beforehand, or perhaps a race somewhere in around an I/O that's supposed to set this variable in a callback if it's asynchronous.  This is the same error that's being reported by the one MacOS user that can't get it to work either.
    The minority's error just reports "Write exception" 2-3 time at exactly the same point.
    it "could" be the case that they're both the exact same error and are just being reported differently,  The last happy-case message that gets printed out is which COM port has been found that's being used.  I will have to admit that I haven't confirmed that it's the one corresponding to their Arduino, because that thought just occurred to me.  But there's really nothing else unusual connected to their laptops, and I have the sense that the Arduino is involved in some I/O because I think that I've seen some of the LEDs on the board twitching during the startup.
    Unfortunately I don't have an ASCII stack trace that I can give you, just a screenshot of the stack trace.  However, it does contain the full trace along with the runtime error from Python.  The image is attached.

     A very few users can get a program to startup every ~10 times that they start it, but the vast majority have never gotten through the constructor call for the startup ever.  One MacOS user has this same problem.

    I haven't had a chance to look at the pymata_core code yet, but I figured that in the interests of parallelism, I'd send this off to you while I look at it myself.  I didn't see much in the way of issues for either "pymata-aio" or "pymata" in GitHub which is where I went to look first, nor did Google turn up anything.

    I have the "unfortunate" situation in that this all works fine on both my laptop and my desktop, so I can't reproduce the problem myself.  The only thing that I've done so far is to swap out your Arduino side for the for the StandardFirmataPlus from the Arduino distribution, which also works for fine (less the features that you've added, of course).  But I haven't had a chance to swap that piece out on a known-failing machine to see if it makes any difference.  However, I don't expect it to, but I thought that it would be a quick-and-easy thing to try.
    My next attempt was to try your non-pymata-aio implementation (PyMata() vs PyMata3()) but I'm having trouble getting my COM port open because it seems to be really wedged at this point.  Before I rebooted and started back along that path I wanted to get this note off to you.

    Have you seen any either same-machine near-but-not-complete startup flakiness, or outright 100% failures on Windows machines, because we do seem to have both.  I'm guessing the root cause is the same though.
    Is there anything else that you would have me try?  At this point I think that we've got everything shoehorned into Anaconda/JupyterLab correctly because everyone gets all the way to the constructor, so all of the gears seem to be in place.  I can say that every time we run our programs we do have to restart the JupyterLab kernel (which I'm assuming just kills and restarts a shared REPL somewhere) every time we have to run a Python program.  But in all cases it *seems* like the COM port is getting closed-up fine for subsequent runs, because that's working fine for half of our (luckier) Windows users.

Thanks very much in advance for any guidance that you may have.

best regards,
-eric

ericwertz, 2018-10-14:
I absolutely want to look at PyCharm.  It's on the top of my PyIDE list, partially due to your references to it.  I originally only saw it in the context of a Chinese embedded board (maybe DFRobot) and assumed that it was a Chinese hack, and until I looked into it didn't realize that it was a JetBrains product.

Our students aren't future-programmers-by-trade so a full-blown IDE is probably too mch for them -- they're pretty much only going to be living in JupyterLab for the near/medium-term future.  But it's what I'll be firing-up the first moment I get the chance.

Thanks for the pointed advice.  Taken!

-e

https://www.jetbrains.com/student/

ericwertz, 2018-10-14:
Here's the screenshot of the stack trace that I've been referring to.  It's not the only error, but is the 90% case.  The other one is the "Write exception" which comes out (possibly through stdout or stderr), in-band, into the output window in JupyterLab.

-e

MrYsLab, 2018-10-14:
Hi Eric,
   Your attachments did not make it through. I think you may have the wrong email address. The correct one is MisterYsLab@gmail.com.

   I am in the Eastern time zone of the U.S.

   If you want a simple IDE you might want to look at the mu-editor https://codewith.mu/.
It is simple, does syntax checking for straight python, allows you to set breakpoints. Only thing is that it is python 3 only.

ericwertz, 2018-10-14:
Alan,
   I did send that follow-up email with the screenshot attachment, and it looks like it went through, but I'll attach it here again just in case.  It's simply the "not a str, but NoneType" error when trying to print out firmware_version in the start-up messages.
   I did since find that you were in NJ since my last message, and that you were "Mister" at Gmail, not "Mr".  I just didn't want to keep sending you email if it were already late and you were in BrexitLand, and felt like you were obligated to respond... 😞
   Feel free to let me know at any point that you're done for the day and I'll figure out what to do next.  Since I can't reproduce the problem myself, I was think that my best approach was to try to understand what's going on in code (async, nice...! 🙁) and probably instrument the sh!t out of Python code so that I can narrow down the exact point and cause of the problem when I get to touch a machine with the problem and swap in a new pymata_core.py file on it and see what else it additionally can tell me.
   I can say with a fair degree of confidence that noone seems to be running off of the end of the hard-coded list of COM ports that only looks through the first twenty.  I would have that problem on my laptop (which has seen ~100 Arduinos recently, and so is up about COM100) but that's not what I'm faced with on half of our students' machines.  I also have no reason to believe that many, if any, students are running less than Windows10.

thanks again,
-e

MrYsLab, 2018-10-14:
I haven't seen your screenshots yet, but some thoughts. Is it possible that there is another serial device that the PC thinks is available? pymata selects the first port that it can connect to so potentially could select the wrong port (but I have no reports of this happening). If you bring up the Arduino IDE and check the COM port it is using, is this the same one that pymata is selecting (close the Arduino IDE before running pymata)? They should be the same.

You may already have checked this out, but this [Arduino page](https://startingelectronics.org/software/arduino/installing-arduino-software-windows-10/) explains getting Arduino to work with Windows 10.
It is possible the driver is the culprit and that page explains how to install an Arduino USB/Serial driver.

I had a user who could not connect his micro:bit to my s2m program using Windows 10. He had to load the appropriate driver to get things to work. Supposedly Windows is already loaded with the correct driver, but in his case, it wasn't. I love Windows ;-).

ericwertz, 2018-10-15:
Alan,
   I believe that I've already done that on at least one machine with the expected results.  As I mentioned, I believe that I've even seen the LEDs twitch on the Arduino side during the initial handshake, so I think the communication channel is intact.  However, I *can't* be 100% sure of this.  On my own machine, I have only the one COM port.  Student's laptops aren't really connected to anything else USB in the lab, and near 100% won't have ever seen another Arduino prior to this, so I'd almost expect them all to come up as COM3.  in fact, I don't remember ever seeing a startup message from any of them that says that PyMata hadn't identified a COM port other than COM3 to use, and I've had to look at ~30 unique instances of this error over the last three lab sessions with all fresh students.

  It is very possible, if not probable, that many students haven't installed the driver that they're prompted to during an Arduino IDE install, because installing the Arduino IDE hasn't been a pre-condition to them getting set up for this semester.  I do know that some of them *have* installed it, so it absolutely is a possibility that this could be the problem, or contributing to it.  This is a great idea.
  It may not explain why a small number of students have the initial PyMata handshake only work 5-10% of the time, but at this point, it's a great line to pursue and I'll have them install the either just the driver or the whole IDE once I take a look to see which is less likely that they'll mess up... 1/2 😊.
   There was a window of time a few years ago in which Windows actually shipped with a competent FTDI driver for use with pre-Unos, but then they switched away from FTDI to the 8u2 (then 16u2).  I didn't think to take a look at their Device Manager to see if the USB ID was being tagged as "Arduino/Genuino Uno" and not just generic serial.  I like this possibility a lot -- not that it'll influence the outcome in any way, of course 😊.
   I'll send out a message to the class and see if anyone pays attention to it today, once I figure out how best to guide them to install the driver.
   I really appreciate your time.  I'll let you know what happens, either way.  Not sure if I'll hear back from anyone by the end of the day or not.

tkx++,
-eric

MrYsLab, 2018-10-15:
Hi Eric,
    I haven't received your screenshots yet and don't see anything attached to this issue. BTW, could you send an email to MisterYsLab@gmail.com. I need to mention something to you, but don't wish to share it publicly here. I was very surprised to see something in my inbox when I woke this morning and would like to comment on it.

If your students are using Windows 7, then I believe they have to install the Arduino drivers. Also, I seem to remember, a long while back, that I had some funny things happen with Windows 10 with the USB/Serial stuff and that all disappeared after one of the many updates. This may not help, but your students should make sure their Windows 10 is up to date.

ericwertz, 2018-10-15:
I sent out an email to all students with non-working setups yesterday telling them to install the IDE and ensuring that the drivers get installed as part of that (the default).  I also asked each of those *with* a working setup if they already installed the IDE previously.  I just work up so I haven't checked to see if I got any responses yet, but I will have in an hour or so when I get to campus, and I'll let you know what I've found out so far.

Again, I appreciate the help.

regards,
-eric

MrYsLab, 2018-10-16:
Here are some error conditions I have caught that might help to isolate the problem if they persist:

I am running blink.py as a test program for 1 & 2.

1. No Arudino plugged in:
```
pymata_aio Version 2.25	Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyS0

Initializing Arduino - Please wait... Traceback (most recent call last):
  File "/home/afy/PycharmProjects/pymata-aio/examples/blink.py", line 29, in <module>
    board = PyMata3()
  File "/home/afy/PycharmProjects/pymata-aio/pymata_aio/pymata3.py", line 68, in __init__
    self.core.start()
  File "/home/afy/PycharmProjects/pymata-aio/pymata_aio/pymata_core.py", line 302, in start
    print("\nArduino Firmware ID: " + firmware_version)
TypeError: can only concatenate str (not "NoneType") to str
```
2. Arduino plugged in but a non-Firmata sketch is loaded on Arduino:

```
pymata_aio Version 2.25	Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... Traceback (most recent call last):
  File "/home/afy/PycharmProjects/pymata-aio/examples/blink.py", line 29, in <module>
    board = PyMata3()
  File "/home/afy/PycharmProjects/pymata-aio/pymata_aio/pymata3.py", line 68, in __init__
    self.core.start()
  File "/home/afy/PycharmProjects/pymata-aio/pymata_aio/pymata_core.py", line 302, in start
    print("\nArduino Firmware ID: " + firmware_version)
TypeError: can only concatenate str (not "NoneType") to str
```
3. Here is a program with signal handling to exit cleanly when the user hits Control-C:
```
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants
import sys
import signal
value = 0

def my_callback(data):
    global value
    # data[0] is the pin number and data[1] is the changed value
    value = (data[1])

try:
    board = PyMata3()
    board.set_pin_mode(2, Constants.ANALOG, my_callback)
except Exception:
    print('ooops')
    raise

# Signal handler to trap control C
def _signal_handler(sig, frame):
    if board is not None:
        print('\nYou pressed Ctrl+C')
        sys.exit(1)

signal.signal(signal.SIGINT, _signal_handler)
signal.signal(signal.SIGTERM, _signal_handler)

# add SIGALRM if platform is not windows
if not sys.platform.startswith('win32'):
    signal.signal(signal.SIGALRM, _signal_handler)

while True:
    print(value)
    board.sleep(.1)
```
and its output:
```
/usr/local/bin/python3.7 /home/afy/PycharmProjects/pymata-aio/examples/comms_loss.py

pymata_aio Version 2.25	Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

0
702
702
702
702
702
702

You pressed Ctrl+C
```
4. Now the same program with control-c handling removed:
```
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants
import sys
value = 0

def my_callback(data):
    global value
    # data[0] is the pin number and data[1] is the changed value
    value = (data[1])

try:
    board = PyMata3()
    board.set_pin_mode(2, Constants.ANALOG, my_callback)
except Exception:
    print('ooops')
    raise

# Signal handler to trap control C
def _signal_handler(sig, frame):
    if board is not None:
        print('\nYou pressed Ctrl+C')
        sys.exit(1)

while True:
    print(value)
    board.sleep(.1)

```
and its output:
```
/usr/local/bin/python3.7 /home/afy/PycharmProjects/pymata-aio/examples/comms_loss.py

pymata_aio Version 2.25	Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

0
702
702
702
702
702
702
702
702
702
702
702
702
702
702
702
702
Traceback (most recent call last):
  File "/home/afy/PycharmProjects/pymata-aio/examples/comms_loss.py", line 26, in <module>
    board.sleep(.1)
  File "/home/afy/PycharmProjects/pymata-aio/pymata_aio/pymata3.py", line 616, in sleep
    self.loop.run_until_complete(task)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 555, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 523, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 1722, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/local/lib/python3.7/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt

```

ericwertz, 2018-10-21:
Alan,
    In the #3 test case that showed above, you're seeing a symptom of a problem that we just discovered in the course of a lesson that I've had students do.
    Notice that the first value returned by the equivalent of analogRead() is returning zero, even though it soon becomes obvious that the steady-state value is nothing near zero.  We've also been finding that the first call to "analogRead" often (if not always) returns zero.  Our workaround has been, of course, to throw away the first sample since we discovered it early yesterday.  Not sure if this problem has an origin outside of just this analogRead symptom, but if Murphy has his (her?) way, there might even be a more generic first-read-of-anytype-or-on-any-pin problem somewhere.  I haven't gotten around to seeing if it also has a digitalRead doppelganger anywhere.

regards,
-eric

  1.  Arduino plugged in but a non-Firmata sketch is loaded on Arduino:

  1.  Here is a program with signal handling to exit cleanly when the user hits Control-C:

  1.  Now the same program with control-c handling removed:

MrYsLab, 2018-10-21:
Eric,
   By adding a small delay before starting the while loop, solves this problem - it give Firmata a chance to complete its first scan of the Ananlog ports.  This is shown below with the call to board.sleep(.04). The output is below the code.

```
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants
import sys
import signal
value = 0

def my_callback(data):
    global value
    # data[0] is the pin number and data[1] is the changed value
    value = (data[1])

try:
    board = PyMata3()
    board.set_pin_mode(2, Constants.ANALOG, my_callback)
except Exception:
    print('ooops')
    raise

# Signal handler to trap control C
def _signal_handler(sig, frame):
    if board is not None:
        print('\nYou pressed Ctrl+C')
        sys.exit(1)

signal.signal(signal.SIGINT, _signal_handler)
signal.signal(signal.SIGTERM, _signal_handler)

# add SIGALRM if platform is not windows
if not sys.platform.startswith('win32'):
    signal.signal(signal.SIGALRM, _signal_handler)

# added to let Firmata get to its first scan
board.sleep(.04)

while True:
    print(value)
    board.sleep(.1)

```

Console output:

```
pymata_aio Version 2.26	Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023
1023

You pressed Ctrl+C

Process finished with exit code 1
```

ericwertz, 2018-10-22:
Yeah, we noticed pretty much the same thing, but we'd just call "analogRead" again.
So, no way to block the caller until the data becomes valid?  Personally, I'd rather even take an exception than bad data if blocking were out of the question.

I vaguely remember pretty much this same thing being a problem in the Arduino library many years ago.

Maybe this is something that I can take a look at myself after we get past our wave of immediate emergencies.

thanks!