MrYsLab/pymata-aio #84: pls bump up PyMata3(arduino_wait=xxx) default value from 2 to >=3

ericwertz, 2018-10-20:
About 40% of our Windows user base is getting crashes (line 302@pymata_core.c where firmware_version is trying to be printed out after being concatenated onto an info string) yielding "NameError: must be str, not NoneType)".

We've found that by overriding the default delay time of two seconds (for the Uno-flavored Arduinos) all of our users can get through this rough spot.  If the default were changed for "arduino_wait" from 2 to at least 3, this could help lots of WIndows users.

thanks!

MrYsLab, 2018-10-20:
No problem. I will also address issues #81 and #83. As soon as I have tested, bumped the revision and placed it on pypi, I will let you know here and close out the issues.

MrYsLab, 2018-10-20:
Eric, you are good to go. I appreciate your efforts in helping to resolve the issues. Resolved with release v 2.26

closed 2018-10-20 by MrYsLab