MrYsLab/pymata-aio #85: using current Arduino IDE, FirmataPlusMrYsLab.ino gives warnings when compiled

ericwertz, 2018-10-21:
When compiling the FirmataPlus library, at least two different (types of) warnings are issued.  It's not clear if they were issued by previous versions of the compiler tools, but they are now.

The first is the unexpected, unadorned URL information that's trying to be a comment after a #include directive.  In particular (but not the only instance of, in FirmataPlusMrYsLab.ino),

#include <AdaEncoder.h> https://code.google.com/p/adaencoder/downloads/detail?name=adaencoder-v0.7beta.zip

I don't ever remember this being legal C, and the current compiler does complain about it.  Changing this to a real //-style comment eliminates this warning.

The other warning, also in FirmataPlusMrYsLab.ino, is:

C:\Users\ericw\Documents\Arduino\FirmataPlusMrYsLab\FirmataPlusMrYsLab.ino: In function 'void sysexCallback(byte, byte, byte*)':

FirmataPlusMrYsLab.ino:849:36: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

           printData("argc = ", argc) ;

                                    ^
This warning can be squelched simply by casting the string to (char *).

ericwertz, 2018-10-21:
Sorry, to clarify -- this is when compiling for AVR.  Unclear what the different flavors of gcc-targets do, but they shouldn't be architecture-related.

MrYsLab, 2018-10-21:
Yes, the toolchain was "updated" along the way and these warning started appearing. Since working with the Firmata sketch is, well,  let us say,  unpleasant, and since these were "just warnings", I decided to leave things as-is, since I knew the code works. I did document the warning in the [WiKi](https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino#special-note).

I will classify the warning into 2 groups. Ones that I introduced and those that are contained in the libraries I am using (the Adafruit encoder library and the oopinintchange library that Adafruit library requires.)

If I ever go into the Firmata sketch code again, I will clean up my warnings since that is simple to do, but I won't be touching the external libraries. The maintainer for these libraries has been unwilling to make any changes. These libraries are quite old and have not been touched in 4 or 5 years. So, since the sketch will produce warnings, no matter what I change in my code, I made this a very low priority item.

When I ran a software group, I always insisted that we have no warnings when publishing code, so I can understand where you are coming from.

MrYsLab, 2018-10-21:
I took a quick look at twarnings, and the level of reporting set in the IDE preferences dialog may open up a Pandora's box of discussion. Taking StandardFirmata as an example, if "Compiler warnings" is set to "All", a whole slew of warnings is issued. So the question (as philosophical as it might be) is, what is the appropriate level of warnings to check for? Why would the "Default" level be the correct one as opposed to any of the other levels? Why isn't a stricter level the correct one and how many libraries could be compiled at stricter levels without some warning be issued?

At the default level, I could easily fix my 3 or 4 warnings, but personally don't think that warrants the effort for a new release. If a user doesn't need the encoder support, they can use the FirmataPlus32u4 sketch which is identifical to FimrataPlus, except it has the encoder support removed. The 32u4 version works on the other supported Arduino boards (Uno, Mega, etc) and was created because at some point, when the AVR toolchain was changed by the Arduino folks, and it no longer could compile the encoder libraries when the Leonardo was selected as the board type. In the past, it didn't complain and I had a single sketch. If the user then tried to use encoder support on a Leonardo, they would always get a value of zero back, and this was documented by me.

So to make a long story short (or longer), I would be willing to change the FirmataPlus code at some point in the future in order to not generate warnings in my code when compiled to a default warning level, but the Adafruit library would still generate warnings.  So I am not sure if these changes warrant the effort in creating a new release, that in fact only "fixes" warnings that have no effect on the operation of the code, and only fixes the warnings generated within the code I have written, but not the code written by others (the Adafruit encoder support).

MrYsLab, 2018-10-22:
An update - I can easily fix the warnings for default level of warnings (I think). However, it appears that for pymata-aio, rotary encoder support is not functioning (but it does for the older PyMata library).
I will open an issue for rotary encoder support.

I will get back to you here when I solve the rotary encoder issue. This may be several days.