MrYsLab/pymata-aio #88: Trying to read from RX

MrYsLab, 2018-11-12:
@Kushal4092  reported the following error:
Hi, I am using Pymata3
https://htmlpreview.github.io/?https://github.com/MrYsLab/pymata-aio/blob/master/documentation/html/pymata3.m.html#header-classes

library and I am trying to read Rx from my arduino but it says I need to upload FirmataPlusRB. when I do that I get an error. When I upload FirmataPlusRBPixy its fine, but when I am trying to use the pixy_get_block I get empty [] and when I wait but a bit longer the program crashes and says
Initializing Arduino - Please wait... Write exception
Write exception
Write exception

Shutting down ...
Write exception

I was using StandardFirmata and It was working fine but I was not able to use the this function.

this is the Code:
board.pixy_init(max_blocks=1, cb=None,cb_type=Constants.CB_TYPE_ASYNCIO) print(board.pixy_get_blocks())
@MrYsLab

MrYsLab, 2018-11-12:
FirmataPlusRBPixy is specifically used when attaching a [PixyCam](https://pixycam.com/pixy-cmucam5/) mounted on a Sparkfun Redbot Robot [https://www.sparkfun.com/products/12649].  FirmataPlusRB is used specifically for the Sparkfun Redbot (but no pixy support).

If you are not using either of those devices, I suggest using FirmataPlus for Uno compatibles and FirmataPlus32u4 for Leonardo compatibles.

Now, on to reading RX. You have not provided much detail about your project, but when Firmata is used with any serial device (USB/Serial, Bluetooth, WiFly, etc.), you do not have access to the RX and TX pins - they are dedicated and cannot be shared. 

Does this answer your question?

Kushal4092, 2018-11-14:
I am trying to get the values from Rx and Tx from another device to my arduino which uses UART programmable interface. I came across Pixy Serial Protocol.