MrYsLab/pymata-aio #91: IndexError with Analog Pin number

yogeshck, 2019-01-02:
I am using Windows 10 with Python 3.5 and Standard Firmata (v2.5.8). I'm using an Arduino based board which has **analog pin A6**. I want to read analog values from this pin **A6**, which is working with Arduino IDE.
But with **pymata-aio** I am getting **IndexError**:
`IndexError: list index out of range`

I tried with this code:

```
import time
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants

board = PyMata3(com_port='COM4')

ldrpin = 6

def my_callback(data):
    global value
    value = (data[1])

def LDR():
    board.set_pin_mode(ldr, Constants.ANALOG, my_callback)
    board.enable_analog_reporting(ldr)
    
    while True:
        print(value)
        board.sleep(1)
```
How can I use pin A6 ?

MrYsLab, 2019-01-02:
When you are working with the Arduino IDE, what board type do you select?

Also, when the program starts up, it states how many digital analog pins have been discovered. What are the values shown in your console?

yogeshck, 2019-01-02:
In Arduino IDE, I'm using **Arduino Duemilanove (ATMEGA328P)**.
It shows: `Found 20 Digital Pins and 6 Analog Pins`
No, A6 is not recognized by pymata-aio.
Thanks for the reply.

MrYsLab, 2019-01-02:
There appears to be some issue with the way that Firmata is recognizing the board type. In the Boards.h file, Firmata is making a compile-time decision on how many analog pins are active:
```
// Arduino Duemilanove, Diecimila, and NG
#if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328__)
#if defined(NUM_ANALOG_INPUTS) && NUM_ANALOG_INPUTS == 6
#define TOTAL_ANALOG_PINS       6
#define TOTAL_PINS              20 // 14 digital + 6 analog
#else
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS 22 // 14 digital + 8 analog
```
If modified the above by commenting out the section to select 6 analog pins and it reports properly now:
```
#if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328__)
#if defined(NUM_ANALOG_INPUTS) && NUM_ANALOG_INPUTS == 6
//#define TOTAL_ANALOG_PINS       6
//#define TOTAL_PINS              20 // 14 digital + 6 analog
//#else
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS              22 // 14 digital + 8 analog```

```
```
Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 22 Digital Pins and 8 Analog Pins
```

I will report this as a bug against Firmata and hopefully they will resolve the issue. Until then, you can modify [Boards.h](https://github.com/MrYsLab/pymata-aio/blob/master/FirmataPlus/source/libraries/FirmataPlus/Boards.h) to force the correct number of pins. If you are using a FirmataPlus type script, each type has its own Boards.h. If you are using StandardFirmata, there is one file shared by all the types.

yogeshck, 2019-01-03:
I tried modifying the board.h, Since I am using Standard Firmata 2.5. I modified the code from the this directory: _C:\Program Files (x86)\Arduino\libraries\Firmata_

closed 2019-01-03 by yogeshck

yogeshck, 2019-01-04:
Its Working. Thank you for the help.
I used FirmataPlus instead of Standard Firmata and its working.
I assume the problem was with administrative permission in windows to modify the  files. 

```
Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 22 Digital Pins and 8 Analog Pins
```

Thank You very much.

MrYsLab, 2019-01-04:
Thanks for letting me know.