MrYsLab/pymata-aio #92: Shutting Down Write Exception

VinealShiloh, 2019-05-07:


MrYsLab, 2019-05-07:
Could you please copy your script and paste it here so I can see exactly what it is doing?

Also, what is QtPiFirmata.ino?

VinealShiloh, 2019-05-08:
**QtPiFirmata.ino** is Modified Firmata of **Standard Firmata** according to our specs.
 We wrote the code for testing our Micro Controller which is Arduino based.The below code is just a snippet to test Analog Read functionality
**CODE SNIPPET**
    #ANALOG READ Method
    def Analog_Read(self):

        resAR = 0

        i = 0
        z = 0

        AnalogReadList_1 = []
        AnalogReadList_2 = []
        AnalogWriteList_1 = []

        #PIN CONFIG for ANALOG PINS (Ports 1-4)
        APin_0 = 0 #Port 1
        APin_1 = 1 #Port 2
        APin_2 = 2 #Port 3
        APin_3 = 6 #Port 4

        #PIN CONFIG for LED PINS (Ports 1-4)
        Led1 = 3
        Led2 = 9
        Led3 = 10
        Led4 = 11

        #Declare Analog Pins
        self.board.set_pin_mode(APin_0, Constants.ANALOG)
        self.board.set_pin_mode(APin_1, Constants.ANALOG)
        self.board.set_pin_mode(APin_2, Constants.ANALOG)
        self.board.set_pin_mode(APin_3, Constants.ANALOG)

        #Declare LED Pins
        self.board.set_pin_mode(Led1, Constants.OUTPUT)
        self.board.set_pin_mode(Led2, Constants.OUTPUT)
        self.board.set_pin_mode(Led3, Constants.OUTPUT)
        self.board.set_pin_mode(Led4, Constants.OUTPUT)

        #Enable Analog reporting (starts reading Analog values)
        self.board.enable_analog_reporting(APin_0)
        self.board.enable_analog_reporting(APin_1)
        self.board.enable_analog_reporting(APin_2)
        self.board.enable_analog_reporting(APin_3)

        #LDR Read 1 (When LEDs are OFF)
        a1= self.board.analog_read(APin_0)
        AnalogReadList_1.append(a1)
        b1= self.board.analog_read(APin_1)
        AnalogReadList_1.append(b1)
        c1= self.board.analog_read(APin_2)
        AnalogReadList_1.append(c1)
        d1= self.board.analog_read(APin_3)
        AnalogReadList_1.append(d1)

        #LEDs ON
        self.board.digital_write(Led1,1)
        self.board.digital_write(Led2,1)
        self.board.digital_write(Led3,1)
        self.board.digital_write(Led4,1)

        self.board.sleep(1)

        #LDR Read 2 (When LEDs are ON)
        a2= self.board.analog_read(APin_0)
        AnalogReadList_2.append(a2)
        b2= self.board.analog_read(APin_1)
        AnalogReadList_2.append(b2)
        c2= self.board.analog_read(APin_2)
        AnalogReadList_2.append(c2)
        d2= self.board.analog_read(APin_3)
        AnalogReadList_2.append(d2)

        #LEDs OFF
        self.board.digital_write(Led1,0)
        self.board.digital_write(Led2,0)
        self.board.digital_write(Led3,0)
        self.board.digital_write(Led4,0)

        print("\nANALOG READ\n")
        for z in range(len(AnalogReadList_1)):
            if AnalogReadList_1[z] < AnalogReadList_2[z]:
                print("Port",z+1,"-True")
                resAR += 1
                self.truthAR.append("TRUE")
            elif AnalogReadList_2[z] <= AnalogReadList_1[z]:
                print("Port",z+1,"-False")
                self.truthAR.append("FALSE")

        print("\n")

        if resAR == 4:
            self.results.append("True")
        elif resAR != 4:
            self.results.append("False")

        print("\n----------------ANALOG READ & WRITE - DONE-----------------\n")

MrYsLab, 2019-05-11:
I am going to close this issue since I have not received example code to recreate it. Most likely your code is trying to send a command to the Arduino after you have called shutdown. If you would like additional help or wish to provide example code, please do that here and I can reopen the issue.

closed 2019-05-11 by MrYsLab