MrYsLab/pymata-aio #95: ULTRASONIC Sensor HC-SR04

nikhilabhi, 2019-06-20:
I am trying use ultrasonic sensor with arduino atmega32. I am using the below code to run the ultrasonic it's giving me a some random value. Can you look at the code and suggest me if am doing wrong some where

**CODE**
from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants
import time
while 1:
    print("Enter COM Port Number:")
    port = input()
    if port != "":
        break

board = PyMata3(com_port='COM'+port)
TRIG = 9
ECHO = 15
board.set_pin_mode(TRIG, Constants.OUTPUT)
board.set_pin_mode(ECHO, Constants.INPUT)

def distance():
    board.digital_write(TRIG, True)
    time.sleep(0.00001)
    board.digital_write(TRIG, False)
    StartTime = time.time()
    StopTime = time.time()
    while board.digital_read(ECHO)==0:
        StartTime = time.time()
    while board.digital_read(ECHO)==1:
        StopTime = time.time()
    TimeElapsed = StopTime - StartTime
    #distance = (TimeElapsed * 34300) / 2
    distance = TimeElapsed * 17150
    distance = round(distance, 2)
    print(distance)
    return distance
while True:
    dist=distance()
    print(dist)
    time.sleep(1)

MrYsLab, 2019-06-20:
I transferred your issue to the pymata-aio repository since your code uses pymata-aio.
In the future, when you post code on Github to maintain its formatting, please place your code between
 "fence" marks (three back ticks). Here is a sample of what I mean:

\```
def print_it(data)
    print(data)
\```

This will look like this:
```
def print_it(data)
    print(data)
```

First, I am assuming that you have [loaded the FirmataPlus script on your Arduino.](https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino).

Have you tried using the [sample program for using the HC-SR04?](https://github.com/MrYsLab/pymata-aio/blob/master/test/ping.py) There is also a [wiring diagram that matches the example.](https://github.com/MrYsLab/pymata-aio/blob/master/documentation/images/ping.png)

closed 2019-06-22 by nikhilabhi

MrYsLab, 2019-06-22:
Thanks for reformatting the code. Your code most likely is not working because you are trying to do "real-time operations" with a non-realtime operating system. Windows, Linux, and MacOS are all non-realtime. As a result, you most likely will not be able to accurately achieve the 10 us timing periods.

Normally one would do this directly on the Arduino (which is what FirmataPlus does) since there is no
operating system on the Arduino to get in the way of the timing.

nikhilabhi, 2019-06-22:
Ok sir. Thanks for your reply.
Can u send me some examples to send sysex commands to firmata using
pymata3

MrYsLab, 2019-06-22:
Pymata does not have a facility for sending custom sysex commands, since it requires the Arduino Firmata sketch to have code to interpret those commands. 

I wrote [an article](https://www.instructables.com/id/Going-Beyond-StandardFirmata-Adding-New-Device-Sup/) several years back on how to modify PyMata (the original library) to add your own sysex commands to both PyMata and the Arduino sketch.

The pymata-aio library is very similar, but the core pymata-aio class is written using Python asyncio.

It may be simpler to start with [PyMata](https://github.com/MrYsLab/PyMata) and if you are successful, then porting to pymata-aio.

From experience, writing the Python side is very straight forward. The complication is trying to debug the Arduino side. The FirmataPlus sketch has a utility to send a sysex message back to either PyMata or pymata-aio and to have the values printed on the Python console. The code on the Arduino side can be [viewed here.](https://github.com/MrYsLab/pymata-aio/blob/70d74368e1f8cf34973b0acb03d5b48ef7a129d8/FirmataPlus/source/libraries/FirmataPlus/examples/FirmataPlus/FirmataPlus.ino#L1131), and the code on the Python side can be [viewed here.](https://github.com/MrYsLab/pymata-aio/blob/70d74368e1f8cf34973b0acb03d5b48ef7a129d8/pymata_aio/pymata_core.py#L1828)

If you have any questions, I would be happy to answer them if I am able.

nikhilabhi, 2019-06-24:
hi sir, thanks for your response towards my msg, I tried sending sysex
commands using Pymata and I was Successful in sending. Now I want to send
Sysex commands using Pymata-aio, But i was unable to send.Can you please
help me out sending sysex command using Pymata aio.

MrYsLab, 2019-06-24:
The python-aio library consists of 2 pieces, pymata_core.py which is written using Python asyncio, and Python3, which is a wrapper around pymata_core for those not familiar with asyncio programming.

First, anything you have written for the Arduino sketch should not have to be changed for pymata-aio.

If you are not familiar with asyncio, you may wish to [read this article.](https://realpython.com/async-io-python/).

If you are adding a new sysex command, that new commands needs to be added to the [private_constants.py](https://github.com/MrYsLab/pymata-aio/blob/master/pymata_aio/private_constants.py) file.

To add a new method to send a sysex command to the Arduino, you can look at [analog_write](https://github.com/MrYsLab/pymata-aio/blob/70d74368e1f8cf34973b0acb03d5b48ef7a129d8/pymata_aio/pymata_core.py#L565) as an example. To have that command be Pymata3 compatible, you can use the [analog_write](https://github.com/MrYsLab/pymata-aio/blob/70d74368e1f8cf34973b0acb03d5b48ef7a129d8/pymata_aio/pymata3.py#L89) in pymata3.py as an example. The parameter list in both the pymata_core and pyamta3 methods must be the same. If you look at the pymata3 methods, they all call pymata_core methods in a very similar manner.

Now If you are expecting data to be returned from the Arduino, you will need to add the sysex command to  the [command_dictionary.](https://github.com/MrYsLab/pymata-aio/blob/70d74368e1f8cf34973b0acb03d5b48ef7a129d8/pymata_aio/pymata_core.py#L111) An entry into this dictionary contains both the value of the sysex command you may have added to private_constants.py and the name of the method that will handle the receipt of the command.