MrYsLab/pymata-aio #96: Installing FirmataPlus on arduino DUE

willmendil, 2019-07-22:
Hi, I'm trying to install firmataplus on arduino due, but neither firmataplus and firmataplus32u4 compile:

`Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

Multiple libraries were found for "Servo.h"
In file included from C:\Users\$USER$\Documents\Arduino\libraries\FirmataPlus\examples\FirmataPlus\FirmataPlus.ino:48:0:

 Used: C:\Users\$USER$\Documents\Arduino\libraries\Servo
C:\Users\$USER$\Documents\Arduino\libraries\ooPinChangeInt/ooPinChangeInt.h:114:20: fatal error: new.h: No such file or directory

 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
    #include <new.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

MrYsLab, 2019-07-22:
The complication is that many of the libraries written for the 328 and 32u4 type processors have not been ported to the SAM which is used by the DUE.

I would first try loading StandardFirmata (not StandardFirmataPlus) and see if  you can get that to compile and work. I do not own a DUE or I would try that here.

If that works, which of the added features of FirmataPlus do you require for your application:
  HC-SRO4 Ultrasonic Distance Sensors using a single pin.
    Stepper Motors.
    Piezo Tone Generation.
    2 Pin Rotary Encoder Support.

If you can identify working DUE libraries for the feature(s) that you need, I might be able to construct a special DUE version of FimrataPlus, but I would need you to test that since, again, I do not own a DUE.

willmendil, 2019-07-22:
The StandardFirmata works fine. I need HCSR04 and servos. as well as custom analog and digital sensors. 
The issue with the StandardFirmata  is that when I run the example here "[https://gist.github.com/MrYsLab/8b735813e413bf62b455](https://gist.github.com/MrYsLab/8b735813e413bf62b455)

i.e. 👍 
```
#!/usr/bin/env python3

from pymata_aio.pymata3 import PyMata3
from pymata_aio.constants import Constants

def pin_6_pwm_128():
    """
    Set digital pin 6 as a PWM output and set its output value to 128
    @return: No Value Returned.
    """
    # instantiate the pymata_core API
    board = PyMata3()

    # set the pin mode
    board.set_pin_mode(6, Constants.PWM)
    board.analog_write(6, 128)

    # wait for 3 seconds to see the LED lit
    board.sleep(3)

    # reset the board and exit
    board.shutdown()

if __name__ == "__main__":
    pin_6_pwm_128()
```

I get (with the native port): 

```
pymata_aio Version 2.30 Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait...
Arduino Firmware ID: None
*** Analog map retrieval timed out. ***

Do you have Arduino connectivity and do you have a Firmata sketch uploaded to the board?

Shutting down ...
Task exception was never retrieved
future: <Task finished coro=<PymataCore._command_dispatcher() done, defined at /home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py:1425> exception=SystemExit(0)>
Traceback (most recent call last):
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 1438, in _command_dispatcher
    next_command_byte = await self.read()
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_serial.py", line 155, in read
    await asyncio.sleep(self.sleep_tune)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 568, in sleep
    return await future
concurrent.futures._base.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 352, in start
    loop.run_until_complete(asyncio.sleep(.1))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 571, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1775, in _run_once
    handle._run()
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 1487, in _command_dispatcher
    await self.shutdown()
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 1231, in shutdown
    sys.exit(0)
SystemExit: 0

```
On the programming port 

```
pymata_aio Version 2.30 Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... _encoder_data() missing 1 required positional argument: 'data'
Shutting down ...

```

willmendil, 2019-07-22:
hope this is related, but trying your firmataexpress, I get: 

```
FirmataExpress:40:21: error: avr/wdt.h: No such file or directory

 #include <avr/wdt.h>

                     ^

compilation terminated.

exit status 1
avr/wdt.h: No such file or directory

```

is it related?

MrYsLab, 2019-07-22:
Servo support and digital and analog input should work with StandardFirmata. The issue is going to be with the HC-SR04. I am using a library called [newping.](https://playground.arduino.cc/Code/NewPing/) To time the pulses on the HC-SR04, it uses timers that are not available on the Due. I am not aware of any HC-SR04 libraries that support the Due. There are Due examples that do their own timing, but that it is not likely to work with Firmata sketches, because of the 10ms loop time of Firmata. You would probably be better off using another type of sensor such as the [Sharp Infrared Proximity Sensor.](https://www.sparkfun.com/products/8958)

The cause of the error you are seeing with FirmataExpress is because the Due does not have a watchdog timer feature and so the compiler cannot find an equivalent function to support the Due.

The USB port that should be used is the [programming port. ](https://www.arduino.cc/en/Guide/ArduinoDue#toc13) I am confused as to how you are seeing the error about _encoder_data. That function is called when the Firmata sketch returns encoder data changes. If you do a hard reset of the Due and then start the Python script, do you still see this error? Do you have any sensors or actuators attached to the Due when you run this script? One cause might be that the Due is sending "junk" out the serial port, and perhaps a reset will stop that from happening if that is the cause.

Could you try running the [blink script](https://github.com/MrYsLab/pymata-aio/blob/master/examples/blink.py) with StandardFirmata and let me know if you are seeing the same error?

Although I doubt this is the error.  You may need to [set the arduino_wait parameter to something like 5 seconds.](https://github.com/MrYsLab/pymata-aio/blob/70d74368e1f8cf34973b0acb03d5b48ef7a129d8/examples/blink.py#L20)

Thanks for your efforts in trying to hunt this down.

willmendil, 2019-07-22:
ok, bad news.
I've been uninstalling arduino completly and reinstling it as well as trying what you said. Here are the results. 

First off, when I upload the standardfirmata on the due I get this output: 

```
In file included from C:\Program Files (x86)\Arduino\libraries\Firmata\examples\StandardFirmata\StandardFirmata.ino:28:0:

C:\Program Files (x86)\Arduino\libraries\Firmata/Firmata.h:42:0: warning: "PWM" redefined [enabled by default]

 #define PWM                     0x03 // same as PIN_MODE_PWM

 ^

In file included from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam3xa/include/sam3xa.h:44:0,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam3.h:59,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam.h:198,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/libsam/chip.h:25,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:42,

                 from sketch\StandardFirmata.ino.cpp:1:

C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:466:0: note: this is the location of the previous definition

 #define PWM        ((Pwm    *)0x40094000U) /**< \brief (PWM       ) Base Address */

 ^

In file included from C:\Program Files (x86)\Arduino\libraries\Firmata\Firmata.cpp:18:0:

C:\Program Files (x86)\Arduino\libraries\Firmata\Firmata.h:42:0: warning: "PWM" redefined [enabled by default]

 #define PWM                     0x03 // same as PIN_MODE_PWM

 ^

In file included from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam3xa/include/sam3xa.h:44:0,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam3.h:59,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam.h:198,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/libsam/chip.h:25,

                 from C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:42,

                 from C:\Program Files (x86)\Arduino\libraries\Firmata\Boards.h:22,

                 from C:\Program Files (x86)\Arduino\libraries\Firmata\Firmata.h:17,

                 from C:\Program Files (x86)\Arduino\libraries\Firmata\Firmata.cpp:18:

C:\Users\$USER$\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:466:0: note: this is the location of the previous definition

 #define PWM        ((Pwm    *)0x40094000U) /**< \brief (PWM       ) Base Address */

 ^

In file included from C:\Program Files (x86)\Arduino\libraries\Firmata\utility\SerialFirmata.h:24:0,

                 from C:\Program Files (x86)\Arduino\libraries\Firmata\utility\SerialFirmata.cpp:20:

C:\Program Files (x86)\Arduino\libraries\Firmata/Firmata.h:42:0: warning: "PWM" redefined [enabled by default]

 #define PWM                     0x03 // same as PIN_MODE_PWM

[... 84 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 352, in start
    loop.run_until_complete(asyncio.sleep(.1))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 571, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1775, in _run_once
    handle._run()
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 1487, in _command_dispatcher
    await self.shutdown()
  File "/home/pi/robotics/testenv/lib/python3.7/site-packages/pymata_aio/pymata_core.py", line 1231, in shutdown
    sys.exit(0)
SystemExit: 0

```

I have tried pressing for a few second the reset button on the Due, as well as unplugging any wires from the due.

What do you want me to try?

willmendil, 2019-07-22:
I tried uploading the FirmataPlus32u4 on my mega2560 and ran the blink.py, 
I mainly get the `Arduino Firmware ID: None` error and gto `_pixy_data() missing 1 required positional argument: 'data'` once ...

MrYsLab, 2019-07-22:
I just tested blink.py on my mega2560 using both Arduino 1.8.8 and the latest 1.8.9 and it is working here. I tested on both Windows 10 and Ubuntu.

I was not able to install the Arduino IDE on windows from the Microsoft Store - it would just hang, so I downloaded the installer and it worked.

You may need to [update the Windows Arduino driver. ](https://www.arduino.cc/en/Guide/DriverInstallation)

Please let me know if that solves the issue.

MrYsLab, 2019-07-22:
For the PWM issue above, I just tried compiling StandardFirmata for the Due here, and I am seeing the same error you are seeing. It looks this is a bug in StandardFirmata. You may wish to enter an issue against [firmata.](https://github.com/firmata/arduino) I would do it, but I have an outstanding issue since January that was not addressed, and perhaps you will have better luck than I. If you prefer that I create the issue, please let me know and I will.

willmendil, 2019-07-22:
On windows it works with arduino 1.8.9 with standardfirmata on the mega. (it did struggle to detect the right port as COM3 is used by intel active management technologuy (on my machine), the arduino was on COM10. )
This also works on the Due. (that's good news).

On the pi, neither work. The mega spites out the `Arduino Firmware ID: None` error and the Due, the `_encoder_data()` or `_pixy_data()` errors. 

The only difference, apart from the OS, is that I'm using python3.7 on the pi and 3.6 on windows. I am uninstalling python3 on the pi  and reinstalling 3.6 to verify if this is the issue. will keep posted. 

regarding the PWM issue, I'll copy the message on firmata. :)

MrYsLab, 2019-07-22:
I just tested my mega using  blink.py on 2 different  Raspberry Pi 3B+ boards, one running  Stretch and the other Buster. The mega is running FirmataPlus32u4 and all is working here.  Stretch is running python 3.6 and Buster 3.7.
I use python 3.7 on both Ubuntu and Windows. 

When things start working strange on my Pis, I have had to do a new install of Raspbian. I am not sure if a corrupt SD card is your issue or not, or perhaps a hardware problem or bad USB cable?

BTW, I do not run the Arduino IDE on the pi, but from my Ubuntu box.

willmendil, 2019-07-22:
ok ... I don't think this will change much, but I'll do what I really didn't want to do and start over with a new SD card and install of raspbian

willmendil, 2019-07-22:
ok after a huge irritating day with getting python3 to work properly (why is python2 still a thing sigh....) anyway, I can attest that the blink works for both the mega and due.  I hoep this is going to be stable

So now, if you want me to try firmataplus for the due, we can do that. Regarding the use of HC-SR04, that's pretty annoying. Isn't there alternatives to the NEWPING? ultrasonic from erick [https://github.com/ErickSimoes/Ultrasonic](https://github.com/ErickSimoes/Ultrasonic) wouldn't work?

MrYsLab, 2019-07-22:
I would be willing to try substituting  Ultrasonic for newping. This would be done as a specific version for the Due. It appears that the tone library is also not supported for the Due, so that will need to be removed.  

What is interesting, is that FirmataPlus is derived from an older version of StandardFirmata (2.5.3), and I am not seeing the compile-time errors that we saw in the latest version of StandardFirmata. 

Have you tried the Utrasonic Library and does it work on the Due?

BTW, I am not willing to substitute Ultrasonic for newping for any of the other versions other than the special Due version, since newping works well and I do not want to introduce any breaking changes.

Of course, I will not be able to test any of this, so it may take several go-arounds. Making changes to the Firmata sketches is never easy because there is almost no easy way to do debugging.

willmendil, 2019-07-22:
Sure, I understand. The fact that you're still maintain these repositories is quite kind enough. 
I'll give a try for the ultrasonic library on the Due, I think it has worked in the past, but I can't remember if I did it on the mega or due. 
Regarding the ultrasonic library, it works fin on the Due. I've tested the code from the example here: 
[https://github.com/ErickSimoes/Ultrasonic/blob/master/examples/MultipleUltrasonicSensors/MultipleUltrasonicSensors.ino](https://github.com/ErickSimoes/Ultrasonic/blob/master/examples/MultipleUltrasonicSensors/MultipleUltrasonicSensors.ino)
```

#include <Ultrasonic.h>

Ultrasonic ultrasonic1(22, 23);  // An ultrasonic sensor HC-04
Ultrasonic ultrasonic2(10);   // An ultrasonic sensor PING)))
Ultrasonic ultrasonic3(8);    // An Seeed Studio ultrasonic sensor

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.print("Sensor 01: ");
  Serial.print(ultrasonic1.read()); // Prints the distance on the default unit (centimeters)
  Serial.println("cm");

  Serial.print("Sensor 02: ");
  Serial.print(ultrasonic2.read(CM)); // Prints the distance making the unit explicit
  Serial.println("cm");

  Serial.print("Sensor 03: ");
  Serial.print(ultrasonic3.read(INC)); // Prints the distance in inches
  Serial.println("inc");

  delay(1000);
}
```

MrYsLab, 2019-07-22:
Ok, I will give it a go. It may take me several days to get to this and will let you know when I have something to test.

MrYsLab, 2019-07-22:
I must be losing it. I used the Ultrasonic library for Firmata-Express. I will just port what I did to FirmataPlus. Hopefully, this will save a lot of time.

MrYsLab, 2019-07-23:
We’ve lost power here due to storms. I was porting when the power went down and was about to compile. Since a significant part of NJ is down, not sure when power will be restored. So if I don’t reply, I am not ignoring you😉

willmendil, 2019-07-23:
[unrelated] hi again, this is an unrelated message to the issue, but was sure how i could reach you. I saw on your blog that you are familiar with zmq. could I direct you to this issue i'm having, posted on stackoverflow. 
[https://stackoverflow.com/questions/57156822/pyzmq-req-rep-with-asyncio-await-for-variable](https://stackoverflow.com/questions/57156822/pyzmq-req-rep-with-asyncio-await-for-variable)

You may delete this message once read

MrYsLab, 2019-07-23:
Still no power. Take a look at python_banyan/projects/OneGPIO/arduino_uno/arduino_gateway.py

MrYsLab, 2019-07-23:
On github

willmendil, 2019-07-23:
I'm sorry, I don't understand your answer :s

MrYsLab, 2019-07-23:
Finally got power up. You can find the Due FirmataPlus version here: https://github.com/MrYsLab/pymata-aio/tree/master/FirmataPlus/source/libraries/FirmataPlusDue
Let me know if this works.

MrYsLab, 2019-07-23:
Sorry about the confusion. I posted a reply to your zmq question on stackoverflow.

willmendil, 2019-07-24:
Ok, sorry for the delay. 

There seems to be an issue as I get: 

```
Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Users\$USER$\Documents\Arduino\libraries\FirmataPlusDue\FirmataPlusDue.cpp:19:24: fatal error: FirmataDue.h: No such file or directory

Multiple libraries were found for "Servo.h"
 #include "FirmataDue.h"

 Used: C:\Users\$USER$\Documents\Arduino\libraries\Servo
                        ^

 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
compilation terminated.

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I though this was a typo in FirmataPlusDue.cpp, so I changed FirmataDue.h with FirmataPlusDue.h but then I get:

```
Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Users\$USER$\Documents\Arduino\libraries\FirmataPlusDue\examples\FirmataPlusDue\FirmataPlusDue.ino: In function 'void systemResetCallback()':

FirmataPlusDue:906:15: error: 'noTone' was not declared in this scope

       noTone(i) ;

               ^

Multiple libraries were found for "Servo.h"
 Used: C:\Users\$USER$\Documents\Arduino\libraries\Servo
 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
exit status 1
'noTone' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I'm sorry, I'm not good enough with c coding to help you out here. Although I think it might be trivial to fix.

MrYsLab, 2019-07-24:
I updated the file on Github.
Weird - compiling in a standalone directory no errors were reported, but moving things into the Arduino/libraries directory, the noTone error appeared. The way the Arduino IDE manages paths is a mess. 

I never see the servo error here, so to simplify things in testing I cleaned out my Arduino/libraries directory to contain only:
```
Arduino
└── libraries
    ├── FirmataPlusDue
    │   ├── Boards.h
    │   ├── examples
    │   │   └── FirmataPlusDue
    │   │       ├── FirmataPlusDue.ino
    │   │       ├── LICENSE.txt
    │   │       └── Makefile
    │   ├── FirmataPlusDue.cpp
    │   ├── FirmataPlusDue.h
    │   ├── keywords.txt
    │   └── LICENSE.txt
    ├── readme.txt
    └── Ultrasonic
        ├── CODE_OF_CONDUCT.md
        ├── CONTRIBUTING.md
        ├── examples
        │   ├── MultipleUltrasonicSensors
        │   │   └── MultipleUltrasonicSensors.ino
        │   ├── Timeout
        │   │   └── Timeout.ino
        │   └── UltrasonicSimple
        │       └── UltrasonicSimple.ino
        ├── extras
        │   ├── HC-SR04-with-Arduino.jpg
        │   └── Ultrasonic.fzz
        ├── keywords.txt
        ├── library.properties
        ├── LICENSE
        ├── README.md
        └── src
            ├── Ultrasonic.cpp
            └── Ultrasonic.h

```
I compiled on both Ubuntu and Windows without errors.  

Let me know if this is still causing errors for you. If so, if you just compile using the checkmark button on the IDE do you get errors and if you upload to the due, does that cause errors?

willmendil, 2019-07-24:
Still have the noTone error both when uploading to the Due and only compiling the code in the Aarduino IDE.  (That's on windows 10)
I've cleared up the library forlder and only installed ultrasonic from erick through the IDE and copied the FirmataPlusDue redownloaded from github:

```
Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Users\$USER$\Documents\Arduino\libraries\FirmataPlusDue\examples\FirmataPlusDue\FirmataPlusDue.ino: In function 'void systemResetCallback()':

FirmataPlusDue:906:15: error: 'noTone' was not declared in this scope

       noTone(i) ;

               ^

exit status 1
'noTone' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

here's the folder tree.

```
libraries
    ├───FirmataPlusDue
    │   │   .gitignore
    │   │   Boards.h
    │   │   FirmataPlusDue.cpp
    │   │   FirmataPlusDue.h
    │   │   keywords.txt
    │   │   LICENSE.txt
    │   │
    │   └───examples
    │       └───FirmataPlusDue
    │               FirmataPlusDue.ino
    │               LICENSE.txt
    │               Makefile
    │
    └───Ultrasonic
        │   CODE_OF_CONDUCT.md
        │   CONTRIBUTING.md
        │   keywords.txt
        │   library.properties
        │   LICENSE
        │   README.md
        │
        ├───examples
        │   ├───MultipleUltrasonicSensors
        │   │       MultipleUltrasonicSensors.ino
        │   │
        │   ├───Timeout
        │   │       Timeout.ino
        │   │
        │   └───UltrasonicSimple
        │           UltrasonicSimple.ino
        │
        ├───extras
        │       HC-SR04-with-Arduino.jpg
        │       Ultrasonic.fzz
        │
        └───src
                Ultrasonic.cpp
                Ultrasonic.h
```

MrYsLab, 2019-07-24:
I updated FirmataPlusDue.cpp earlier. It looks like you might not have picked up the changes. The lines around 906 were commented out.

Let me know if you are still seeing compile errors.

willmendil, 2019-07-24:
Are you sure you commented the right file? in the [FirmataPlusDue.ino](https://github.com/MrYsLab/pymata-aio/blob/master/FirmataPlus/source/libraries/FirmataPlusDue/examples/FirmataPlusDue/FirmataPlusDue.ino) it is not commented and last edit was 16 hours ago. 
In the [FirmataPlusDue](https://github.com/MrYsLab/pymata-aio/tree/master/FirmataPlus/source/libraries/FirmataPlusDue) rep, only the FiremataPlusDue.ccp file was edited a hour ago and it was the import typo

MrYsLab, 2019-07-24:
Sorry, I did not push the file. It should work now.

willmendil, 2019-07-24:
SUCCESS !! I tried the blink on the raspberry pi and it works. I also tryed the [ping.py](https://github.com/MrYsLab/pymata-aio/blob/master/test/ping.py) example and it works fine also. you're the boss

closed 2019-07-24 by willmendil