MrYsLab/pymata-aio #97: Pymata_aio call back

nikhilabhi, 2019-09-03:
Sir, I am using pymata aio and firmata. I want to return some data from arduino firmata to Python idle. Whenever i send sysex command it should return me back with data (Callback), But this is not happening with pymata aio. Can u please help to fix this issue?

MrYsLab, 2019-09-03:
Here is an [example of using callbacks](https://github.com/MrYsLab/pymata-aio/blob/master/examples/simple_analog_input_with_callback.py).

If you are still having problems, please post your code and I will take a look to see what might be the issue.

MrYsLab, 2019-09-04:
I am closing this issue, but if you have any questions you can post here and if necessary I can reopen the issue.

closed 2019-09-04 by MrYsLab