MrYsLab/pymata-aio #100: Firmata Plus is not compiling on arduino nano

AlessandroDotta, 2019-09-27:
Hi There,

I follow your procedures for the installation, including all the libraries, but the following errors are still showing up when I try to compile the FirmataPlus Sketch:

> 
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':

(.text+0x0): multiple definition of `__vector_7'

libraries\NewPing\NewPing.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.
> 

Thanks for your help.

MrYsLab, 2019-09-27:
I just tried compiling for the nano using a clean install of the Arduino IDE (1.8.9 on Ubuntu and 1.8.10 on Windows) and then added the FirmataPlus files to the library and I am not seeing the error. 

Here is a screenshot of my Arduino directory on Windows (it is the same on Ubuntu):
[image]

If you are not sure where your Arduino directory is located, if you open the Arduino IDE and then click File and Preferences you should see something like this. The path to the file should be shown.
[image]

Does your libraries directory look like the one shown above?

MrYsLab, 2019-09-29:
I am closing this issue. If you are still having issues, you can post here, and I will see the post even though the issue is now closed. If need be, I will reopen the issue.

closed 2019-09-29 by MrYsLab