MrYsLab/pymata-aio #104: asyncio.set_event_loop_policy(asyncio.WindowsSelectorEventLoopPolicy()) Error

abixadamj, 2020-02-03:
Hi,
I use pymata-aio with my library to connect Arduino.
On Windows 10 frome some time (personally I use Linux) I see an error:
```

    self.board = PyMata3(com_port=port)
  File "C:\Users\Szkolenie\AppData\Local\Programs\Python\Python36\lib\site-packages\pymata_aio\pymata3.py", line 71, in __init__
    asyncio.set_event_loop_policy(asyncio.WindowsSelectorEventLoopPolicy())
AttributeError: module 'asyncio' has no attribute 'WindowsSelectorEventLoopPolicy'
```

It has not been few releses back. I am using 3.6.8 version o Python for Windows, and 3.6.9 version of Python for Linux. On Linux everything is OK. Can anybody help me? Google does not know anything about such error...

MrYsLab, 2020-02-03:
Hi Adam,
    Would you be able to either upgrade Windows to Python 3.7 or Python 3.8 or install the previous release of pymata-aio? Other than supporting WindowsSelectorEventLoopPolicy, there are no other differences between pymata-aio v2.34 and 2.33.

To install the previous version of pymata-aio first uninstall the current version:

```
pip uninstall pymata-aio
```

To install v2.33:

```
pip install pymata-aio==2.33
```

MrYsLab, 2020-02-05:
Were you able to get things working on Windows?

MrYsLab, 2020-02-06:
This issue was addressed with version 2.35. Windows support is now limited to python 3.7 and above. You can also use the suggestions made above if you need to use Python 3.5 or Python 3.6 on Windows.

closed 2020-02-06 by MrYsLab