MrYsLab/pymata-aio #105: Callback never getting called using pymata-core

Petapton, 2020-03-17:
Hello, I can't manage to get a program to work, since when I set a callback it is never called.
Although I can clearly see the Tx pin blinking and the callback is properly set, my PymataCore object doesn't seem to react.
What's going up? What am I doing wrong?
Thank you.

MrYsLab, 2020-03-17:
Without seeing your code, it is very difficult to diagnose your issue. 

Here is a working example for a digital input with callback:
```
import asyncio

from pymata_aio.pymata_core import PymataCore
from pymata_aio.constants import Constants

async def the_callback(data):
    print(data)

async def pin_12_digital_input(my_board):

    # set the pin mode
    await board.set_pin_mode(12, Constants.INPUT, the_callback, Constants.CB_TYPE_ASYNCIO)

    # keep the program alive for 20 seconds
    await asyncio.sleep(20)

    # shutdown
    await my_board.shutdown()

# create a PyMataCore instance and complete the initialization with a call to start()
board = PymataCore()
board.start()

loop = asyncio.get_event_loop()
loop.run_until_complete(pin_12_digital_input(board))

```

Please be aware that in the next few weeks, I will be archiving this repository. Since you are using pymata-core, I suggest you use [pymata-express](https://mryslab.github.io/pymata-express/). It provides a pure asyncio API that is simpler to use than pymata-aio, and that has many working examples. You can view an example for digital input with callback [here](https://github.com/MrYsLab/pymata-express/blob/master/examples/digital_input.py), or analog input with callback [here](https://github.com/MrYsLab/pymata-express/blob/master/examples/analog_input.py).
The PyMata repository is also going to be archived and replaced with pymata4. Pymata4 uses threading as its concurrency model. I am finishing up on the documentation for pymata4, and as soon as it is ready, I will be moving things over to archive.

Petapton, 2020-03-17:
Yes, thank you. I just discovered pymata express, but I'm having issues with it also. I'll try to solve them tomorrow(as long as trying your snippet), otherwise I will open an issue for pymata express. Thank you again.

MrYsLab, 2020-03-18:
I am going to close this issue. If necessary, I can reopen it. If you have any questions or comments you may enter them here and I will see them even though the issue is closed.

closed 2020-03-18 by MrYsLab